openbmc/openbmc #3268: Not getting the service “xyz.openbmc_project.Control.FanSpeed” for Fan Control

gayathril2703, 2018-06-22:
I have checked for the service “xyz.openbmc_project.Control.FanSpeed” using busctl tool.

In Fan control code they are using this service , we also have that service yaml file in the build directory at location build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-dbus-interfaces/1.0-r1/git/xyz/openbmc_project/Control/FanSpeed.interface.yaml .

But I checked it on server side , I didn’t get that service over the server. 
Also I have checked status of that service using busctl as well as systemctl tool also but that service itself is not available.

[image]

After that using busctl tree command , I listed out all the running services on the server with its respective tree. In that it is not present. 

I also checked for the busctl get property command , but that is also getting failed because service is unavailable.

[image]
 
And busctl introspect also getting failed.

[image]

**The config used for building is meta-Olympus which is 2400 based platform.**

I saw that, the same case is working for Romulus config.
Do I need to make any changes in the yaml files or need to create the service?

Please someone help to resolve this.

mine260309, 2018-06-25:
Which machine is it?

Please be noted that OpenBMC has two interfaces of fan control, `FanSpeed` and `FanPwm`.
It depends on the hwmon config to choose which interface to use.
E.g. Witherspoon uses `FanSpeed` (by default) and Romulus uses `FanPwm`, by [fan monitor config](https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/meta-ibm/meta-romulus/recipes-phosphor/fans/phosphor-fan-monitor-config-native/monitor.yaml#L9) and [fan control config](https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/meta-ibm/meta-romulus/recipes-phosphor/fans/phosphor-fan-control-fan-config-native/fans.yaml#L6)

For details, please refer to https://github.com/openbmc/docs/blob/master/sensor-architecture.md#other-interfaces

spinler, 2018-12-11:
Dormant issue.  Closing.

closed 2018-12-11 by spinler