openbmc/openbmc #3274: Build error using Debian testing in libfdt

bluecmd, 2018-06-26:
Hi,

I'm attempting to build OpenBMC on Debian testing.
This is the command that fails:
```
ocp-bluecmd:~/openbmc/build/tmp/work/x86_64-linux/u-boot-mkimage-native/1_2017.09-r0/git$ gcc  -isystem/home/dhtech/openbmc/build/tmp/work/x86_64-linux/u-boot-mkimage-native/1_2017.09-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/dhtech/openbmc/build/tmp/work/x86_64-linux/u-boot-mkimage-native/1_2017.09-r0/recipe-sysroot-native/usr/lib -L/home/dhtech/openbmc/build/tmp/work/x86_64-linux/u-boot-mkimage-native/1_2017.09-r0/recipe-sysroot-native/lib -Wl,-rpath-link,/home/dhtech/openbmc/build/tmp/work/x86_64-linux/u-boot-mkimage-native/1_2017.09-r0/recipe-sysroot-native/usr/lib -Wl,-rpath-link,/home/dhtech/openbmc/build/tmp/work/x86_64-linux/u-boot-mkimage-native/1_2017.09-r0/recipe-sysroot-native/lib -Wl,-rpath,/home/dhtech/openbmc/build/tmp/work/x86_64-linux/u-boot-mkimage-native/1_2017.09-r0/recipe-sysroot-native/usr/lib -Wl,-rpath,/home/dhtech/openbmc/build/tmp/work/x86_64-linux/u-boot-mkimage-native/1_2017.09-r0/recipe-sysroot-native/lib -Wl,-O1 -Wp,-MD,tools/.atmelimage.o.d -Wall -Wstrict-prototypes -O2 -fomit-frame-pointer    -DCONFIG_FIT_SIGNATURE -include ./include/libfdt_env.h -idirafterinclude -idirafter./arch/sandbox/include -I./lib/libfdt -I./tools -DUSE_HOSTCC -D__KERNEL_STRICT_NAMES -D_GNU_SOURCE  -c -o tools/atmelimage.o tools/atmelimage.c
In file included from ./tools/../lib/libfdt/libfdt.h:10:0,
                 from tools/../include/libfdt.h:1,
                 from tools/fdt_host.h:11,
                 from tools/imagetool.h:24,
                 from tools/atmelimage.c:8:
/usr/include/libfdt_env.h:70:30: error: conflicting types for ‘fdt64_t’
 typedef uint64_t FDT_BITWISE fdt64_t;
                              ^~~~~~~
In file included from <command-line>:0:0:
././include/libfdt_env.h:19:16: note: previous declaration of ‘fdt64_t’ was here
 typedef __be64 fdt64_t;
                ^~~~~~~
In file included from ././include/libfdt_env.h:12:0,
                 from <command-line>:0:
/usr/include/libfdt_env.h:90:24: error: expected ‘)’ before ‘x’
 static inline uint32_t fdt32_to_cpu(fdt32_t x)
                        ^
././include/compiler.h:71:9: error: expected ‘)’ before ‘&’ token
  ((((x) & 0xff000000) >> 24) | \
         ^
././include/compiler.h:71:23: error: expected ‘)’ before ‘>>’ token
  ((((x) & 0xff000000) >> 24) | \
                       ^
././include/compiler.h:71:30: error: expected ‘)’ before ‘|’ token
  ((((x) & 0xff000000) >> 24) | \
                              ^
/usr/include/libfdt_env.h:94:23: error: expected ‘)’ before ‘x’
 static inline fdt32_t cpu_to_fdt32(uint32_t x)
                       ^
././include/compiler.h:71:9: error: expected ‘)’ before ‘&’ token
  ((((x) & 0xff000000) >> 24) | \
         ^
././include/compiler.h:71:23: error: expected ‘)’ before ‘>>’ token
  ((((x) & 0xff000000) >> 24) | \
                       ^
././include/compiler.h:71:30: error: expected ‘)’ before ‘|’ token
  ((((x) & 0xff000000) >> 24) | \
                              ^
/usr/include/libfdt_env.h:99:24: error: expected ‘)’ before ‘x’
 static inline uint64_t fdt64_to_cpu(fdt64_t x)
                        ^
././include/compiler.h:76:9: error: expected ‘)’ before ‘&’ token
  ((((x) & 0xff00000000000000##sfx) >> 56) | \
         ^
././include/compiler.h:76:36: error: expected ‘)’ before ‘>>’ token
  ((((x) & 0xff00000000000000##sfx) >> 56) | \
                                    ^
././include/compiler.h:76:43: error: expected ‘)’ before ‘|’ token
  ((((x) & 0xff00000000000000##sfx) >> 56) | \
                                           ^
/usr/include/libfdt_env.h:103:23: error: expected ‘)’ before ‘x’
 static inline fdt64_t cpu_to_fdt64(uint64_t x)
                       ^
././include/compiler.h:76:9: error: expected ‘)’ before ‘&’ token
  ((((x) & 0xff00000000000000##sfx) >> 56) | \
         ^
././include/compiler.h:76:36: error: expected ‘)’ before ‘>>’ token
  ((((x) & 0xff00000000000000##sfx) >> 56) | \
                                    ^
././include/compiler.h:76:43: error: expected ‘)’ before ‘|’ token
  ((((x) & 0xff00000000000000##sfx) >> 56) | \
                                           ^
```

GCC version:
```
ocp-bluecmd:~/openbmc/build/tmp/work/x86_64-linux/u-boot-mkimage-native/1_2017.09-r0/git$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 7.3.0-23' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.3.0 (Debian 7.3.0-23) 
```

See https://patchwork.openembedded.org/patch/147816/ for more discussions and various patches.

bluecmd, 2018-06-26:
My solution to work around the issue is to `sudo apt-get remove libfdt-dev` to force u-boot to use the internal libfdt.

amboar, 2018-06-28:
Interesting. Thanks for posting the work-around.

closed 2019-04-26 by stale[bot]