openbmc/openbmc #3276: Intermittent bitbake download issues

geissonator, 2018-06-27:
We all seem to hit this at some point when doing a bitbake and a new download is needed.

The package varies, depending on what's new, and it's very intermittent.  A rebuild fixes it.  Need to look into a possible retry in bitbake when we hit this type fail?

```
13:28:49 [968A[J[1;31mERROR[0m: [31mslpd-lite-1.0-r1 do_fetch: Fetcher failure: Fetch command export PATH="/home/fspcibld/master-merge-ubuntu-witherspoon/openbmc/scripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/slpd-lite/1.0-r1/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/slpd-lite/1.0-r1/recipe-sysroot/usr/bin/crossscripts:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/slpd-lite/1.0-r1/recipe-sysroot-native/usr/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/slpd-lite/1.0-r1/recipe-sysroot-native/usr/bin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/slpd-lite/1.0-r1/recipe-sysroot-native/sbin:/tmp/openbmc/work/armv6-openbmc-linux-gnueabi/slpd-lite/1.0-r1/recipe-sysroot-native/bin:/home/fspcibld/master-merge-ubuntu-witherspoon/openbmc/import-layers/yocto-poky/bitbake/bin:/tmp/openbmc/hosttools"; export HOME="/home/fspcibld"; LANG=C git -c core.fsyncobjectfiles=0 fetch -f --prune --progress git://github.com/openbmc/slpd-lite refs/*:refs/* failed with exit code 1, output:
13:28:49 remote: Counting objects: 8, done.        
13:28:49 remote: Compressing objects:  25% (1/4)           
remote: Compressing objects:  50% (2/4)           
remote: Compressing objects:  75% (3/4)           
remote: Compressing objects: 100% (4/4)           
remote: Compressing objects: 100% (4/4), done.        
13:28:49 remote: Total 8 (delta 4), reused 8 (delta 4), pack-reused 0        
13:28:49 fatal: missing blob object '004bc8867ca127385c74f49ae58f9dbe926ac196'
13:28:49 error: git://github.com/openbmc/slpd-lite did not send all necessary objects
13:28:49 
13:28:49 [0m
13:28:49 [1;31mERROR[0m: [31mslpd-lite-1.0-r1 do_fetch: Fetcher failure for URL: 'git://github.com/openbmc/slpd-lite'. Unable to fetch URL from any source.[0m
13:28:49 [1;31mERROR[0m: [31mslpd-lite-1.0-r1 do_fetch: Function failed: base_do_fetch[0m
13:28:49 [1;31mERROR[0m: [31mLogfile of failure stored in: /tmp/openbmc/work/armv6-openbmc-linux-gnueabi/slpd-lite/1.0-r1/temp/log.do_fetch.22378[0m
13:28:49 Currently 11 running tasks (4637 of 5638)  82% ||
```

geissonator, 2018-06-28:
More from a build today:
```
ERROR: network-1.0-r1 do_unpack: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/network/1.0-r1/recipe-sysroot-native/usr/bin/python-native:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/scripts:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/network/1.0-r1/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/network/1.0-r1/recipe-sysroot/usr/bin/crossscripts:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/network/1.0-r1/recipe-sysroot-native/usr/sbin:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/network/1.0-r1/recipe-sysroot-native/usr/bin:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/network/1.0-r1/recipe-sysroot-native/sbin:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/network/1.0-r1/recipe-sysroot-native/bin:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/import-layers/yocto-poky/bitbake/bin:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/hosttools"; export HOME="/gsa/ausgsa/home/a/n/andrewg"; git -c core.fsyncobjectfiles=0 clone -s -n /esw/san5/openbmc/dl-mirror//git2/github.com.openbmc.phosphor-networkd/ /tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/network/1.0-r1/git/ failed with exit code 128, output:
Cloning into '/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/network/1.0-r1/git'...
done.

fatal: missing blob object 'c4e989a7c92c0a5cfb8c7a0d534babba4eae1b88'
fatal: remote did not send all necessary objects
fatal: The remote end hung up unexpectedly

ERROR: network-1.0-r1 do_unpack: Function failed: base_do_unpack
ERROR: Logfile of failure stored in: /tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/network/1.0-r1/temp/log.do_unpack.16342
ERROR: Task (/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/meta-phosphor/common/recipes-phosphor/network/network.bb:do_unpack) failed with exit code '1'
ERROR: phosphor-settings-manager-1.0-r1 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-settings-manager/1.0-r1/recipe-sysroot-native/usr/bin/python-native:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/scripts:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-settings-manager/1.0-r1/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-settings-manager/1.0-r1/recipe-sysroot/usr/bin/crossscripts:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-settings-manager/1.0-r1/recipe-sysroot-native/usr/sbin:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-settings-manager/1.0-r1/recipe-sysroot-native/usr/bin:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-settings-manager/1.0-r1/recipe-sysroot-native/sbin:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-settings-manager/1.0-r1/recipe-sysroot-native/bin:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/import-layers/yocto-poky/bitbake/bin:/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/hosttools"; export HOME="/gsa/ausgsa/home/a/n/andrewg"; LANG=C git -c core.fsyncobjectfiles=0 fetch -f --prune --progress git://github.com/openbmc/phosphor-settingsd refs/*:refs/* failed with exit code 1, output:
error: Could not read d4fef299abc972805aec733fab51640707463695
remote: Counting objects: 9, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 9 (delta 2), reused 5 (delta 2), pack-reused 4
fatal: missing blob object '55f0b001c6f7152b6677e6280ea4bc3ea124039c'
error: git://github.com/openbmc/phosphor-settingsd did not send all necessary objects

ERROR: phosphor-settings-manager-1.0-r1 do_fetch: Fetcher failure for URL: 'git://github.com/openbmc/phosphor-settingsd'. Unable to fetch URL from any source.
ERROR: phosphor-settings-manager-1.0-r1 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /tmp/andrewg/2018-06-28_openbmc_daily/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-settings-manager/1.0-r1/temp/log.do_fetch.12528
ERROR: Task (/tmp/andrewg/2018-06-28_openbmc_daily/openbmc/meta-phosphor/common/recipes-phosphor/settings/phosphor-settings-manager.bb:do_fetch) failed with exit code '1'
WARNING: cross-localedef-native-2.27-r0 do_fetch: Failed to fetch URL git://github.com/kraj/localedef;branch=master;name=localedef;destsuffix=git/localedef, attempting MIRRORS if available
```

geissonator, 2018-06-28:
We don't see these on our openpower.xyz nodes so I'm assuming this is probably something with our internal IBM network.  Be curious if anyone else hits it though.

gtmills, 2019-01-23:
@geissonator Can we close?

closed 2019-08-24 by stale[bot]

Ppatilvvdn, 2022-12-13:
I am also getting same error please help to fix it