openbmc/openbmc #3278: No event log when temperature over threshold

LindarLiu0724, 2018-06-28:
Hi all,
I have added a temperature sensor in hwmon, and from the REST interface I notice 
"""WarningAlarmLow = 1"""
but no any event log found. I didn't do any porting for SEL part.

Could somebody give suggestion how to debug and how to make the sel function work?
thanks.

gtmills, 2018-07-05:
Hi @LindarLiu0724, I believe we don't create event logs when sensors are in a warning or critical state. On Witherspoon, we create an event log if 3 or more cores are overtemp using D-Bus monitoring. 
https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/recipes-phosphor/fans/witherspoon-thermal-policy/thermal-policy.yaml#L170
 @spinler could help answer this though. This is a good question for the mailing list, as it's not a bug or feature request. You can sign yourself up at for the list at https://lists.ozlabs.org/listinfo/openbmc

Thanks, 
Gunnar

gtmills, 2018-07-05:
Similiar question on https://github.com/openbmc/openbmc/issues/1828. Maybe @spinler you can close both after answering. Thanks!

spinler, 2018-07-13:
@tomjoseph83 Are there any IPMI stories planned to create SEL events when the various thresholds are crossed?

tomjoseph83, 2018-08-24:
@spinler We do not have any stories open to track mapping the thresholds to SEL entries.

closed 2019-04-26 by stale[bot]