openbmc/openbmc #3282: Witherspoon kernel build broken

shenki, 2018-06-29:
When bumping the kernel witherspoon is failing with this message:

```
ERROR: linux-aspeed-4.17.3+gitAUTOINC+7d98f10b63-r0 do_kernel_metadata: Could not generate configuration queue for aspeed.
ERROR: linux-aspeed-4.17.3+gitAUTOINC+7d98f10b63-r0 do_kernel_metadata: Function failed: do_kernel_metadata (log file is located at /home/joel/dev/obmc/openbmc/build/tmp/work/witherspoon-openbmc-linux-gnueabi/linux-aspeed/4.17.3+gitAUTOINC+7d98f10b63-r0/temp/log.do_kernel_metadata.29840)
ERROR: Logfile of failure stored in: /home/joel/dev/obmc/openbmc/build/tmp/work/witherspoon-openbmc-linux-gnueabi/linux-aspeed/4.17.3+gitAUTOINC+7d98f10b63-r0/temp/log.do_kernel_metadata.29840
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_kernel_metadata
| /home/joel/dev/obmc/openbmc/build/tmp/work/witherspoon-openbmc-linux-gnueabi/linux-aspeed/4.17.3+gitAUTOINC+7d98f10b63-r0/temp/run.do_kernel_metadata.29840: 204: /home/joel/dev/obmc/openbmc/build/tmp/work/witherspoon-openbmc-linux-gnueabi/linux-aspeed/4.17.3+gitAUTOINC+7d98f10b63-r0/temp/run.do_kernel_metadata.29840: spp: not found
| /home/joel/dev/obmc/openbmc/build/tmp/work/witherspoon-openbmc-linux-gnueabi/linux-aspeed/4.17.3+gitAUTOINC+7d98f10b63-r0/temp/run.do_kernel_metadata.29840: 211: /home/joel/dev/obmc/openbmc/build/tmp/work/witherspoon-openbmc-linux-gnueabi/linux-aspeed/4.17.3+gitAUTOINC+7d98f10b63-r0/temp/run.do_kernel_metadata.29840: kgit: not found
| /home/joel/dev/obmc/openbmc/build/tmp/work/witherspoon-openbmc-linux-gnueabi/linux-aspeed/4.17.3+gitAUTOINC+7d98f10b63-r0/temp/run.do_kernel_metadata.29840: 217: /home/joel/dev/obmc/openbmc/build/tmp/work/witherspoon-openbmc-linux-gnueabi/linux-aspeed/4.17.3+gitAUTOINC+7d98f10b63-r0/temp/run.do_kernel_metadata.29840: scc: not found
| ERROR: Could not generate configuration queue for aspeed.
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_kernel_metadata (log file is located at /home/joel/dev/obmc/openbmc/build/tmp/work/witherspoon-openbmc-linux-gnueabi/linux-aspeed/4.17.3+gitAUTOINC+7d98f10b63-r0/temp/log.do_kernel_metadata.29840)
ERROR: Task (/home/joel/dev/obmc/openbmc/meta-openbmc-bsp/meta-aspeed/recipes-kernel/linux/linux-aspeed_git.bb:do_kernel_metadata) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/joel/dev/obmc/openbmc/meta-openbmc-bsp/meta-aspeed/recipes-kernel/linux/linux-aspeed_git.bb:do_kernel_metadata
```

shenki, 2018-06-29:
This is happening with: https://gerrit.openbmc-project.xyz/c/11319/

I don't know what this failure is due to. I suspect it's something to do with Witherspoon's out of tree patches. Can someone from that team please look at this?

mdmillerii, 2018-06-29:
did you have #3281?

anoo1, 2018-06-29:
Thanks Milton. After rebasing, getting now:

```
| DEBUG: Executing shell function do_patch
| (1/1) 0001-ARM-dts-Aspeed-Witherspoon-Update-BMC-partitioning.patch
| [INFO]: check of .kernel-meta//patches//./0001-ARM-dts-Aspeed-Witherspoon-Update-BMC-partitioning.patch with "git am" did not pass, trying reduced context.
| [INFO]: Context reduced git-am of .kernel-meta//patches//./0001-ARM-dts-Aspeed-Witherspoon-Update-BMC-partitioning.patch with "git am" did not work, trying "apply".
```

Eddie will be updating this out-of-tree patch.

amboar, 2018-07-09:
Why is that patch out of tree? Can we not rework things so that the kernel tree properly reflects Witherspoon's requirements? What is preventing that?

I think we should do better than just rebasing that hack.

anoo1, 2018-07-20:
Eddie refreshed the patch. Don't remember why this layout was not submitted upstream, maybe something to bring up in IRC or mailing list.

closed 2018-07-20 by anoo1