openbmc/openbmc #3283: SEL support in OpenBMC

vinodhinij, 2018-06-29:
Hi,
Can anyone please let me know if SEL implementation is available in OpenBMC. Can we add and get sel entries.
I tried adding SEL, but, not able to see the added sel when giving the getsel list via ipmitool.

Regards,
J. Vinodhini

gtmills, 2018-07-05:
Hi @vinodhinij, this is a good question for the mailing list, as it's not a bug or feature request. You can sign yourself up at for the list at https://lists.ozlabs.org/listinfo/openbmc. @tomjoseph83 Might be able to answer this. 

Thanks,
Gunnar

vinodhinij, 2018-07-06:
Hi Gunnar,
Am not able to sign up in the list. Can you please help me out or can I mail directly from this mail ID to tomjoseph ?

Regards,
J. Vinodhini

amboar, 2018-07-09:
> Am not able to sign up in the list.

What went wrong? Can you please provide more detail? It shouldn't be more complicated than filling out the form, submitting, and then responding to the confirmation email (or clicking the embedded link).

tomjoseph83, 2018-07-09:
When the error logging was designed for the OpenBMC platform, it was decided that there will not be separate repositories for IPMI(SEL), RedFish etc. We came up with https://github.com/openbmc/phosphor-logging and decided that SEL records will be generated from the logging entries in the D-Bus. https://github.com/openbmc/phosphor-host-ipmid can be looked up to understand the current implementation.

This implementation is limited to show only H/W callouts in the SEL descriptively. All other logging entries are shown as system events.

Since there is no repository for SEL, Add SEL command also has a limited scope. In the current implementation on the OpenBMC, it is used as a way to translate host firmware SEL entries to D-Bus logging entries.

This falls short of the way the IPMI specification expects it, partly because of the conscious decision taken not to design OpenBMC around IPMI.

HoangDNguyen, 2018-07-13:
Hi Tom,

I tried to use "ipmitool sel list" command, and always see the error type is "Undetermined system hardware failure".
And the value of "associations" property in the https://${bmc}/xyz/openbmc_project/logging/entry/<id> interface always be empty.

It's limitation of OpenBMC?

gtmills, 2018-08-20:
@tomjoseph83 Can you answer @HoangDNguyen question above?

tomjoseph83, 2018-08-24:
@HoangDNguyen Yeah the "association" is needed to map to the correct FRU that is having the issue. If there are no associations, we put the generic sensor/offset for "Undetermined system hardware failure".

Definitely welcome to come out with a better way to map the D-Bus error log to IPMI SEL's.

jmbills, 2018-08-28:
I have been working on a proof-of-concept to move the IPMI SEL entries out of D-Bus into journald instead.

Since journald allows custom metadata for log entries, I've thought of having the SEL message logged to the journal and using metadata to store the necessary IPMI info associated with the entry.  Here is an example of logging a type 0x02 system event entry to journald:
```
sd_journal_send("MESSAGE=%s", message.c_str(),
                            "PRIORITY=%i", selPriority,
                            "MESSAGE_ID=%s", selMessageId,
                            "IPMI_SEL_RECORD_ID=%d", recordId,
                            "IPMI_SEL_RECORD_TYPE=%x", selSystemType,
                            "IPMI_SEL_GENERATOR_ID=%x", genId,
                            "IPMI_SEL_SENSOR_PATH=%s", path.c_str(),
                            "IPMI_SEL_EVENT_DIR=%x", assert,
                            "IPMI_SEL_DATA=%s", selDataStr,
                            NULL);
```
Using journald should allow for scaling to more SEL entries which should also enable us to support more generic IPMI behavior such as the Add SEL command.

gtmills, 2018-08-29:
@vinodhinij Can this be closed now? Thanks!

gtmills, 2019-01-23:
No response. Closing.

closed 2019-01-23 by gtmills