openbmc/openbmc #3285: psuedo cannot build

shenki, 2018-07-02:
the pusedo step failed on my machine:

```
ERROR: pseudo-native-1.9.0+gitAUTOINC+fddbe854c9-r0 do_unpack: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export https_proxy="http://proxy:3128"; export http_proxy="http://proxy:3128"; export PATH="/scratch/joel/openbmc/scripts/native-intercept:/scratch/joel/openbmc/scripts:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin/x86_64-linux:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/bin:/scratch/joel/openbmc/import-layers/yocto-poky/bitbake/bin:/scratch/joel/openbmc/build/tmp/hosttools"; export HOME="/home/joel"; git -c core.fsyncobjectfiles=0 checkout -B master fddbe854c9db058d5a05830d3bcdd4233d95ee2e failed with exit code 128, output:
fatal: reference is not a tree: fddbe854c9db058d5a05830d3bcdd4233d95ee2e

ERROR: pseudo-native-1.9.0+gitAUTOINC+fddbe854c9-r0 do_unpack: Function failed: base_do_unpack
ERROR: Logfile of failure stored in: /scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/temp/log.do_unpack.2603
ERROR: Task (virtual:native:/scratch/joel/openbmc/meta/recipes-devtools/pseudo/pseudo_git.bb:do_unpack) failed with exit code '1'
ERROR: prelink-native-1.0+gitAUTOINC+05aeafd053-r0 do_unpack: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export https_proxy="http://proxy:3128"; export http_proxy="http://proxy:3128"; export PATH="/scratch/joel/openbmc/scripts/native-intercept:/scratch/joel/openbmc/scripts:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/prelink-native/1.0+gitAUTOINC+05aeafd053-r0/recipe-sysroot-native/usr/bin/x86_64-linux:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/prelink-native/1.0+gitAUTOINC+05aeafd053-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/prelink-native/1.0+gitAUTOINC+05aeafd053-r0/recipe-sysroot-native/usr/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/prelink-native/1.0+gitAUTOINC+05aeafd053-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/prelink-native/1.0+gitAUTOINC+05aeafd053-r0/recipe-sysroot-native/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/prelink-native/1.0+gitAUTOINC+05aeafd053-r0/recipe-sysroot-native/bin:/scratch/joel/openbmc/import-layers/yocto-poky/bitbake/bin:/scratch/joel/openbmc/build/tmp/hosttools"; export HOME="/home/joel"; git -c core.fsyncobjectfiles=0 checkout -B cross_prelink 05aeafd053e56356ec8c62f4bb8f7b95bae192f3 failed with exit code 128, output:
fatal: reference is not a tree: 05aeafd053e56356ec8c62f4bb8f7b95bae192f3

ERROR: prelink-native-1.0+gitAUTOINC+05aeafd053-r0 do_unpack: Function failed: base_do_unpack
ERROR: Logfile of failure stored in: /scratch/joel/openbmc/build/tmp/work/x86_64-linux/prelink-native/1.0+gitAUTOINC+05aeafd053-r0/temp/log.do_unpack.2838
ERROR: Task (virtual:native:/scratch/joel/openbmc/meta/recipes-devtools/prelink/prelink_git.bb:do_unpack) failed with exit code '1'
```

```
DEBUG: Executing python function do_unpack
DEBUG: Executing python function base_do_unpack
DEBUG: Running export PSEUDO_DISABLED=1; export https_proxy="http://proxy:3128"; export http_proxy="http://proxy:3128"; export PATH="/scratch/joel/openbmc/scripts/native-intercept:/scratch/joel/openbmc/scripts:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin/x86_64-linux:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/bin:/scratch/joel/openbmc/import-layers/yocto-poky/bitbake/bin:/scratch/joel/openbmc/build/tmp/hosttools"; export HOME="/home/joel"; git -c core.fsyncobjectfiles=0 clone -s -n /scratch/joel/openbmc/build/downloads/git2/git.yoctoproject.org.pseudo/ /scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/git/
DEBUG: Running export PSEUDO_DISABLED=1; export https_proxy="http://proxy:3128"; export http_proxy="http://proxy:3128"; export PATH="/scratch/joel/openbmc/scripts/native-intercept:/scratch/joel/openbmc/scripts:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin/x86_64-linux:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/bin:/scratch/joel/openbmc/import-layers/yocto-poky/bitbake/bin:/scratch/joel/openbmc/build/tmp/hosttools"; export HOME="/home/joel"; git -c core.fsyncobjectfiles=0 remote set-url origin git://git.yoctoproject.org/pseudo
DEBUG: Running export PSEUDO_DISABLED=1; export https_proxy="http://proxy:3128"; export http_proxy="http://proxy:3128"; export PATH="/scratch/joel/openbmc/scripts/native-intercept:/scratch/joel/openbmc/scripts:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin/x86_64-linux:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/bin:/scratch/joel/openbmc/import-layers/yocto-poky/bitbake/bin:/scratch/joel/openbmc/build/tmp/hosttools"; export HOME="/home/joel"; git -c core.fsyncobjectfiles=0 checkout -B master fddbe854c9db058d5a05830d3bcdd4233d95ee2e
ERROR: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export https_proxy="http://proxy:3128"; export http_proxy="http://proxy:3128"; export PATH="/scratch/joel/openbmc/scripts/native-intercept:/scratch/joel/openbmc/scripts:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin/x86_64-linux:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/usr/bin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/sbin:/scratch/joel/openbmc/build/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+fddbe854c9-r0/recipe-sysroot-native/bin:/scratch/joel/openbmc/import-layers/yocto-poky/bitbake/bin:/scratch/joel/openbmc/build/tmp/hosttools"; export HOME="/home/joel"; git -c core.fsyncobjectfiles=0 checkout -B master fddbe854c9db058d5a05830d3bcdd4233d95ee2e failed with exit code 128, output:
fatal: reference is not a tree: fddbe854c9db058d5a05830d3bcdd4233d95ee2e

DEBUG: Python function base_do_unpack finished
DEBUG: Python function do_unpack finished
ERROR: Function failed: base_do_unpack
```

```
import-layers/yocto-poky/meta/recipes-devtools/pseudo/pseudo_git.bb:SRCREV = "fddbe854c9db058d5a05830d3bcdd4233d95ee2e"
```

This commit does exist.

shenki, 2018-07-02:
This was something strange going on with my local build directory. The git repository didn't contain the latest commits.

mine260309, 2018-07-04:
I hit this issue as well.
Clean and rebuild fixes it; but if I do build again, the issue occurs again.

gtmills, 2018-07-05:
Same issue as https://lists.yoctoproject.org/pipermail/poky/2018-May/011356.html ? Might need to read a few "previous messages"

mine260309, 2018-07-06:
So it indicates that Yocto 2.5 requires a clean tree to build. Otherwise it gets such error.
I did hit this issue when I have previous builds.
Probably we can close this issue.

closed 2019-10-18 by stale[bot]