openbmc/openbmc #3286: Phosphor-fan-presence: Cannot use "getPropertyValue" to get temp WarningHigh property through dbus

ThaiThu, 2018-07-04:
Hi all,

We are modifying some action in fan control. (phosphor-fan-presence)
When we use getPropertyValue function we can get the temperature value from dbus:
object: /xyz/openbmc_project/sensors/temperature/test_temp1
interface: xyz.openbmc_project.Sensor.Value
property: Value

But when we also use "getPropertyValue" to get Temperature's WarningHigh we always get "out_of_range" error.
object: /xyz/openbmc_project/sensors/temperature/test_temp1
interface:　xyz.openbmc_project.Sensor.Threshold.Warning
property: WarningHigh

The config is as follow:

SetSpeedEvent{
Group{
{
"/xyz/openbmc_project/sensors/temperature/test_temp1",
{"xyz.openbmc_project.Sensor.Threshold.Warning",
"WarningHigh"}
},
},
std::vector{
make_action(action::count_state_before_speed(
static_cast<size_t>(1),
static_cast<int64_t>(100),
static_cast<uint64_t>(100))
),
},
Timer{
static_caststd::chrono::seconds(0),
util::Timer::TimerType::repeating
},
std::vector{
Signal{
match::propertiesChanged(
"/xyz/openbmc_project/sensors/temperature/test_temp1",
"xyz.openbmc_project.Sensor.Value"
),
make_handler(
propertySignal<int64_t>(
"/xyz/openbmc_project/sensors/temperature/test_temp1",
"xyz.openbmc_project.Sensor.Value",
"Value",
handler::setProperty<int64_t>(
"/xyz/openbmc_project/sensors/temperature/test_temp1",
"xyz.openbmc_project.Sensor.Value",
"Value"
))
)
},
}

Thank you all !!

msbarth, 2018-07-05:
Do you see the `xyz.openbmc_project.Sensor.Threshold.Warning` interface and `WarningHigh` property on the dbus object using the `busctl` application on the BMC?

Also, please use markdown when adding any formatted output(such as code snippets), it helps to format what's entered here so its easily readable.

ThaiThu, 2018-07-06:
hi @msbarth 

Yes, I see `xyz.openbmc_project.Sensor.Threshold.Warning` interface and `WarningHigh` on dbus by using "busctl" command

Thank you !!

msbarth, 2018-07-07:
Ok. Is the code snippet you provided above exactly what was generated within the `phosphor-fan-presence/control` application for this property or was it manually edited? Reason I ask is that seems to have been edited in some way since the propertiesChanged match and handler sections are referring to:
```
"/xyz/openbmc_project/sensors/temperature/test_temp1",
"xyz.openbmc_project.Sensor.Value",
"Value"
```
even though the group is:
```
"/xyz/openbmc_project/sensors/temperature/test_temp1",
{"xyz.openbmc_project.Sensor.Threshold.Warning",
"WarningHigh"}
```

spinler, 2018-07-19:
Closing since there's no updates.  Hopefully @msbarth's last comment cleared it up.

closed 2018-07-19 by spinler