openbmc/openbmc #3287: IPMI FRU list must not show non-present FRUs

AlexanderAmelkin, 2018-07-04:
Currently, phosphor-inventory-manager caches all created nodes forever and doesn't provide a method to remove an inventory item once added.

When the inventory is accessed via REST that poses not much of a problem because of the `Present` property that indicates the actual presence of an inventory component.

However, the IPMI specification doesn't provide a presence attribute for FRU Devices per se.
There are two possible approaches to mitigating this nuisance:

1. Add discrete Presence sensors associated with each FRU Device (users will have to check sensor state to see whether or not a particular FRU is actually present).
2. Add handling of the inventory manager Presence property into IPMI FRU command handlers so that missing FRUs are not reported at all.

Although the former option seems to be more IPMI-compliant, I'm personally inclined towards the latter option as more user-friendly. However, counter arguments are welcome.

edtanous, 2018-09-05:
Most non-openbmc systems I'm aware of use the SDR to modify the type 18 "fru locator" records.  If the component is present, a record is available to the SDR.  If the component is not present, the record is removed.  This way, available fru devices can be scanned without ever returning an error.

Adding logic to IPMI to check the present field and present the correct records would be my preference.

Ramkosh, 2019-04-10:
In work

AlexanderAmelkin, 2019-10-21:
@Ramkosh, don't we have a solution yet?

Ramkosh, 2019-10-21:
Right, I'll push it to gerrit.

Ramkosh, 2020-04-24:
In review.

AlexanderAmelkin, 2020-05-06:
https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-host-ipmid/+/31669

closed 2020-05-11 by geissonator