openbmc/openbmc #3288: Combine template tests where possible

micwalsh, 2018-07-06:
In many of our testcase files, we have multiple template testcases with just one row of data though they all call they same keyword.  This makes it almost not worth coding them as template test cases:

Example:

Set Valid SOL Retry Count
    [Documentation]  Verify valid SOL's retry count via IPMI.
    [Tags]  Set_Valid_SOL_Retry_Count
    [Template]  Verify SOL Setting

    # Setting name    Min valid value    Max valid value
    retry-count       0                  7

Set Valid SOL Retry Interval
    [Documentation]  Verify valid SOL's retry interval via IPMI.
    [Tags]  Set_Valid_SOL_Retry_Interval
    [Template]  Verify SOL Setting

    # Setting name    Min valid value    Max valid value
    retry-interval    0                  255

'Set Valid SOL Retry Count' more easily have been coded this way:

Set Valid SOL Retry Count
    [Documentation]  Verify valid SOL's retry count via IPMI.
    [Tags]  Set_Valid_SOL_Retry_Count

    Verify SOL Setting  retry-count  0  7

Or better yet, make a more generic test name and include several lines of template fields:

Verify SOL Settings
    [Documentation]  Verify different SOL settings
    [Tags]  Verify_SOL_Settings
    [Template]  Verify SOL Setting

    # Setting name    Min valid value    Max valid value
    retry-count       0                  7
    retry-interval    0                  255
    character-accumulate-level  1                  255
    character-send-threshold    0                  255

Examples taken from tests/ipmi/test_sol.robot.

micwalsh, 2018-07-06:
Created in wrong repo.

closed 2018-07-06 by micwalsh

gkeishin, 2018-07-06:
https://github.com/openbmc/openbmc-test-automation/issues/1414