openbmc/openbmc #3293: ERROR: python-gevent-1.0.1-r0 do_compile: python setup.py build execution failed.

yyaocsyy, 2018-07-10:
Hi All,

When I try to build openbmc for meta-centriq2400-rep on a ARM64 host, I get a built error, it seems due to the python-gevent 1.0.1, which is really old.I'm trying to replace it with gevent 1.3.4. Do you have any idea ofr this?

Here is my build command:
`TEMPLATECONF=meta-openbmc-machines/meta-arm/meta-qualcomm/meta-centriq2400-rep/conf . openbmc-env`
 `bitbake obmc-phosphor-image`

The error log is:

> ERROR: python-gevent-1.0.1-r0 do_compile: python setup.py build execution failed.
ERROR: python-gevent-1.0.1-r0 do_compile: Function failed: do_compile (log file is located at /home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/temp/log.do_compile.39696)
ERROR: Logfile of failure stored in: /home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/temp/log.do_compile.39696
Log data follows:
| DEBUG: Executing shell function do_compile
| running build
| running build_py
| running build_ext
| configure: WARNING: using cross tools not prefixed with host triplet
| /home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/gevent-1.0.1/libev/config.guess: line 1450: /tmp/cgXGnOjS/dummy: cannot execute binary file
| /home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/gevent-1.0.1/libev/config.guess: unable to guess system type
|
| This script, last modified 2008-01-23, has failed to recognize
| the operating system you are using. It is advised that you
| download the most up to date version of the config scripts from
|
|   http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.guess;hb=HEAD
| and
|   http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.sub;hb=HEAD
|
| If the version you run (/home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/gevent-1.0.1/libev/config.guess) is already up to date, please
| send the following data and any information you think might be
| pertinent to <config-patches@gnu.org> in order to provide the needed
| information to handle your system.
|
| config.guess timestamp = 2008-01-23
|
| uname -m = aarch64
| uname -r = 4.14.15-6.hxt.aarch64
| uname -s = Linux
| uname -v = #1 SMP Tue Apr 10 08:55:00 UTC 2018
|
| /usr/bin/uname -p = aarch64
| /bin/uname -X     =
|
| hostinfo               =
| /bin/universe          =
| /usr/bin/arch -k       =
| /bin/arch              = aarch64
| /usr/bin/oslevel       =
| /usr/convex/getsysinfo =
|
| UNAME_MACHINE = aarch64
| UNAME_RELEASE = 4.14.15-6.hxt.aarch64
| UNAME_SYSTEM  = Linux
| UNAME_VERSION = #1 SMP Tue Apr 10 08:55:00 UTC 2018
| configure: error: cannot guess build type; you must specify one
| Running '/bin/sh /home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/gevent-1.0.1/libev/configure > configure-output.txt' in /home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/gevent-1.0.1/build/temp.linux-aarch64-2.7/libev
| building 'gevent.core' extension
| arm-openbmc-linux-gnueabi-gcc -march=armv6 -marm -mtune=arm1176jz-s --sysroot=/home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/recipe-sysroot -DNDEBUG -g -O3 -Wall -Wstrict-prototypes -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0=/usr/src/debug/python-gevent/1.0.1-r0 -fdebug-prefix-map=/home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/recipe-sysroot= -fdebug-prefix-map=/home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/recipe-sysroot-native= -fPIC -DLIBEV_EMBED=1 -DEV_COMMON= -DEV_CLEANUP_ENABLE=0 -DEV_EMBED_ENABLE=0 -DEV_PERIODIC_ENABLE=0 -Ibuild/temp.linux-aarch64-2.7/libev -Ilibev -I/home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/recipe-sysroot/usr/include/python2.7 -c gevent/gevent.core.c -o build/temp.linux-aarch64-2.7/gevent/gevent.core.o
| In file included from gevent/libev.h:2:0,
|                  from gevent/gevent.core.c:313:
| libev/ev.c:45:12: fatal error: config.h: No such file or directory
|  #  include "config.h"
|             ^~~~~~~~~~
| compilation terminated.
| error: command 'arm-openbmc-linux-gnueabi-gcc' failed with exit status 1
| ERROR: python setup.py build execution failed.
| WARNING: /home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/temp/run.do_compile.39696:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at /home/yyaocs/work/openbmc_new/build/tmp/work/armv6-openbmc-linux-gnueabi/python-gevent/1.0.1-r0/temp/log.do_compile.39696)
ERROR: Task (/home/yyaocs/work/openbmc_new/import-layers/meta-virtualization/recipes-devtools/python/python-gevent_1.0.1.bb:do_compile) failed with exit code '1'

yyaocsyy, 2018-07-12:
The root cause is the config.guess(config.guess timestamp = 2008-01-23) in python-gevent-1.0.1 is too old.
Which the python-gevent-1.0.1 was release at Apr 30, 2014.
So I think we'd better update the python-gevent to 1.2.2 or newer version.

rfrandse, 2018-07-19:
<https://gerrit.openbmc-project.xyz/11439> gevent: update 1.0.1 -> 1.2.2
Resolves  #3293 ERROR: python-gevent-1.0.1-r0 do_compile: python setup.py build execution failed.

bradbishop, 2018-08-21:
This should be fixed by: https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc/+/11876/

closed 2018-08-21 by bradbishop