openbmc/openbmc #3299: Witherspoon: Forwarding to syslog missed 85 messages.

gkeishin, 2018-07-11:
```
Jul 11 15:17:06 W54 systemd-journald[1709]: Forwarding to syslog missed 85 messages.
```
Looks like in newer code some setting setup journald to forward to syslog and it’s in the systemd recipe `meta/recipes-core/systemd/systemd_237.bb`
```
# Enable journal to forward message to syslog daemon
sed -i -e 's/.*ForwardToSyslog.*/ForwardToSyslog=yes/' ${D}${sysconfdir}/systemd/journald.conf
```

gkeishin, 2018-07-11:
Fix  https://gerrit.openbmc-project.xyz/#/c/11444/

geissonator, 2018-07-11:
http://git.yoctoproject.org/cgit/cgit.cgi/meta-virtualization/tree/recipes-devtools/python/python-gevent_1.0.1.bb?h=master was the commit that changed this default on us.

rfrandse, 2018-07-11:
<https://gerrit.openbmc-project.xyz/11444> Disable forwarding of journal to syslog
Resolves  #3299 Witherspoon: Forwarding to syslog missed 85 messages.

closed 2018-07-24 by geissonator