openbmc/openbmc #3301: rest-server: incorrect error message in do_put

mine260309, 2018-07-12:
In phosphor-rest-server `do_put()`, there is code to handle the case that the default type does not match the interface, and it tries to get the type signature, convert the type, and try to `do_put()` again:
```python
    def do_put(self, path, prop, value=None, retry=True):
        if value is None:
            value = request.parameter_list

        prop, iface, properties_iface = self.get_host_interface(
            path, prop, request.route_data['map'][path])
        try:
            properties_iface.Set(iface, prop, value)
        except ValueError as e:
            abort(400, str(e))
        except dbus.exceptions.DBusException as e:
            if e.get_dbus_name() == DBUS_INVALID_ARGS and retry:
                bus_name = properties_iface.bus_name
                expected_type = get_type_signature_by_introspection(self.bus,
                                                                    bus_name,
                                                                    path,
                                                                    prop)
                if not expected_type:
                    abort(403, "Failed to get expected type: %s" % str(e))
                converted_value = None
                try:
                    converted_value = convert_type(expected_type, value)
                    self.do_put(path, prop, converted_value, False)
                    return
                except Exception as ex:
                    abort(403, "Failed to convert %s to type %s" %
                          (value, expected_type))
                abort(403, str(e))
            raise
```

The exception after the retry says `Failed to convert %s to type %s`, however, the exception may occur in `do_put()` instead of `convert_type()`, then the error message is incorrect.

E.g. if user sets host time via REST API, while it is not allowed due to the time config, DBus will get Insufficient permission error, but REST API reports `Failed to convert xxxto type t`.
```
$ curl -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data":1531378019960321}' https://$bmc/xyz/openbmc_project/time/host/attr/Elapsed
{
  "data": {
    "description": "Failed to convert 1531378019960321 to type t"
  },
  "message": "403 Forbidden",
  "status": "error"
}
```

The expected behavior is that the description correctly reports the error.

mine260309, 2018-07-12:
https://gerrit.openbmc-project.xyz/#/c/11462/

rfrandse, 2018-07-12:
<https://gerrit.openbmc-project.xyz/11462> Fix error message in do_put()
Resolves:  #3301 rest-server: incorrect error message in do_put

closed 2018-07-18 by geissonator