openbmc/openbmc #3306: Phosphor-User-Management:- LDAP Authentication

ratagupt, 2018-07-13:
Support LDAP Authentication on BMC.
We have the following external interfaces
1) ssh 2) rest 3) IPMI

Do we want to support the LDAP authentication for all or partial?

in the case of 1) and 3) we need to write the pam module

ratagupt, 2018-07-16:
@rthomaiy ^^^

rthomaiy, 2018-07-16:
Yes, authentication has to happen through PAM only. But we can use openldap, and update the sssd configuration accordingly, so that it can take care. 
Please have a look at https://mapr.com/support/s/article/How-to-configure-LDAP-client-by-using-SSSD-for-authentication-on-CentOS?language=en_US   (we will be configuring in similar way)

AlexanderAmelkin, 2019-01-15:
When do you think LDAP authentication will be implemented and how exactly it is going to be configurable? Are there plans to add WebUI configuration for it? How are the phosphor user roles going to be bound to LDAP? Using LDAP groups?

sivassrr, 2019-07-17:
With master 2.7 release (2.7.0-dev-1040-g616087d)  able to setup secure LDAP and able to login as well.
Also tried to power on / power off and was able to do successfully with secured LDAP user.
 
For more details check https://github.com/openbmc/openbmc/issues/3578
 
With regards,
Sivas

closed 2020-02-15 by stale[bot]