openbmc/openbmc #3307: bump version as a commit description

bazhenov-dn, 2018-07-13:
Hi everyone,

Currently it is quite frustrating to see the enormous amount of commits with 'bump version' description for various OpenBMC components without any glue o what actually has been changed.

Is it possible to replicate the 'bump version' description with the one from a corresponding component commit to understand changes that the distribution underwent?

Thanks,
Dmitry

spinler, 2018-07-26:
I think that's a good idea.  The autobump script is https://github.com/openbmc/openbmc-tools/blob/master/infra/autobump.pl if you care to take a crack at it.

gtmills, 2018-08-14:
+1 this idea.

spinler, 2018-08-30:
FYI, this was done.  Check out how  https://gerrit.openbmc-project.xyz/#/c/openbmc/meta-phosphor/+/12217/ looks

bazhenov-dn, 2018-08-31:
Looks awesome.

closed 2018-08-31 by bazhenov-dn