openbmc/openbmc #3311: openpower-update-manager: App logs ELOOP error

anoo1, 2018-07-16:
During a pnor update, the openpower-update-manager app catches sdbusplus error
```
sd_bus_call: System.Error.ELOOP: Too many levels of symbolic links
```
when calling Delete from deleteImageManagerObject(). This error doesn't affect the code update operation.

This issue is to find the cause of the error msg and fix it.

mine260309, 2019-01-23:
Some investigation:
* When a PNOR is *deleted* by the `xyz.openbmc_project.Object.Delete` interface, this *delete* effectively calls `ItemUpdater::erase()`;
* The object itself (e.g. `/xyz/openbmc_project/software/966365dc`) remains on the bus, and is managed by ObjectMapper, e.g.
```
	# mapper get-service /xyz/openbmc_project/software/966365dc
	xyz.openbmc_project.ObjectMapper

	# busctl introspect xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/software/966365dc | cat
	NAME                                TYPE      SIGNATURE RESULT/VALUE FLAGS
	org.freedesktop.DBus.Introspectable interface -         -            -
	.Introspect                         method    -         s            -
	org.freedesktop.DBus.Peer           interface -         -            -
	.GetMachineId                       method    -         s            -
	.Ping                               method    -         -            -
	org.freedesktop.DBus.Properties     interface -         -            -
	.Get                                method    ss        v            -
	.GetAll                             method    s         a{sv}        -
	.Set                                method    ssv       -            -
	.PropertiesChanged                  signal    sa{sv}as  -            -
```

So why is the object left on bus and owned by `xyz.openbmc_project.ObjectMapper` service?

mine260309, 2019-01-23:
A simple workaround could be to start a systemd unit to invoke `Delete` method instead of directly invoke it in its own process.

mine260309, 2019-01-23:
But I have a question, why `deleteImageManagerObject()` needs to be called on `finishActivation()`?
After the activation is finished, if the object is to be deleted, the PNOR itself will be deleted as well, is it?
^ @anoo1

spinler, 2019-01-23:
Yea, removing associations in the mapper doesn't work so well now.  I have a fix coming soon.

mine260309, 2019-01-24:
> Please don't do this. Instead, just call the delete handler directly instead of delete.

@bradbishop Sure, schedule a method call to delete in event loop should work better.
But now I do not understand why it needs to be deleted here, need @anoo1's feedback.

> Yea, removing associations in the mapper doesn't work so well now. I have a fix coming soon.

@spinler Good to know! With the fix, https://github.com/openbmc/openpower-pnor-code-mgmt/issues/1 should have been fixed as well.

mine260309, 2019-01-24:
@anoo1 I think I get it why it needs to call delete there in `deleteImageManagerObject()`, it is actually intended to call `xyz.openbmc_project.Software.Version`'s `Delete`, which deletes the temporary files in `/tmp/images`.

At the time `deleteImageManagerObject()` is called, there are two services implementing `xyz.openbmc_project.Object.Delete`:
* `xyz.openbmc_project.Software.Version` which is the service to manage the image in `/tmp/images`
* `org.open_power.Software.Host.Updater` which is the service to manage the PNOR on flash.

So the code needs to *pick* the service instead of using the *first* service returned by mapper.

anoo1, 2019-07-26:
Fixed by https://github.com/openbmc/openpower-pnor-code-mgmt/commit/c9caf86988b5695c117894df1628cba0cc4b4210 but the "Resolves:" was misspelled so it didn't automatically get closed. Closing.

closed 2019-07-26 by anoo1