openbmc/openbmc #3312: NGINX Authentication

geissonator, 2018-07-16:
Currently, nginx requires the proxies behind it do all user authentication.  As we add more proxies behind nginx (REST, Redfish), customers would want a central authentication.

This implies we need to either get rid of nginx and move to a bmcweb front end type implementation (a single server running on the https port that handles authentication for everything behind it) or we need to look into nginx capabilities for providing a single point of authentication.  Certain read-only queries (like root redfish api) need to be whitelisted (i.e. no authentication required). nginx has some support for this type of thing but investigation certainly required.

A lot of this was discussed with @edtanous in IRC today.

edtanous, 2018-08-13:
An nginx auth module was the first path I went down for trying to get all this stuff to work.  I quickly abandoned it, but that's not to say it's not technically possible.  The reasons I didn't pursue it were:

1. Writing nginx modules is comparatively difficult compared to Crow/Beast request interceptors.
2. While there exists some other projects that might help, like OpenResty https://openresty.org/en/ those require the author to know Lua.  If we went down that path (which includes some prewritten modules) it would likely be OpenBMCs first use of Lua, which I wasn't prepared to push from a project perspective.
3. Redfish has weird login requirements around whitelisting resources that go beyond just the URL level.  For example, SessionService should only allow POST when unauthenticated, but not GET.  Getting that level of fine grained control on nginx seemed difficult, especially when adding PrivilegeRegistry, which allows runtime modification of the user Privileges.
4. On the lame-o-meter, having an extra socket, parse, and buffer copy when running nginx in reverse proxy mode seems very lame, especially on a single core processor where we have to context switch to manage it.  In practice, it's not a significant increase in CPU usage compared to pulling the resources from dbus, but it is measurable, non-zero, and affects every API, including the ones that just host static files.  While the lame-o-meter only exists in my head, I like to think other people have them as well.

From a project perspective, it gives users a GREAT ability to customize, as it allows people to boot their own web servers behind the reverse proxy without having to really worry about security.  I think that's somewhat valuable, but I think getting a solution that works and is secure is more important at this point, but others may disagree with me.

gtmills, 2019-01-23:
@geissonator IBM is moving away from nginx, can we close?

closed 2019-08-24 by stale[bot]