openbmc/openbmc #3314: Build framework may produce undetected corrupt images

nest1ing, 2018-07-17:
Building from scratch on some of our machines gives unbootable firmware images.
In qemu it looks like this:
```
U-Boot 2016.07 (Jul 17 2018 - 14:20:27 +0000)

DRAM:  240 MiB
WARNING: Caches not enabled
Flash: 32 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20080000 ...
Bad FIT kernel image format!
ERROR: can't get kernel image!
ast# 
```

At the same time, the same build on another machine may give a correct image.
The build process runs in a docker container identical for all build machines, so all the build machines have the same environment. The only difference is, as we think, is the number of CPUs and cores on those machines.

As we found out, the reason is the size of `fitImage-obmc-phosphor-initramfs-*.bin`. 
It is greater than allowed by `flash-layout` in DT.

We compared `obmc-phosphor-initramfs-*.rootfs.cpio.lzma` from good and bad builds and found that they also have different sizes. We then uncompressed them and compared the "source" cpio archives 
(`obmc-phosphor-initramfs-*rootfs.cpio`). The sizes matched, but the binary contents differed heavily due to, as we believe, different file order in the source file system due to different timing of the builds (esp. taking in account the different number of cores on the build machines).

We think it's a fault in the build scripts that the size of the firmware image parts is not checked agains the flash layout. We believe such a check must be added and the whole build must fail if this check fails.

nest1ing, 2018-07-17:
I post changes to gerrit https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/11516

rfrandse, 2018-07-17:
<https://gerrit.openbmc-project.xyz/11516> Check size of the firmware image parts
Resolves:  #3314 Build framework may produce undetected corrupt images

closed 2018-07-25 by geissonator