openbmc/openbmc #3317: npm install ERR! openbmc phosphor-webui do_compile

timruggles, 2018-07-19:
Unable solve the access problem.
Fails in do_compile with : /local/truggles/hosttools/git ls-remote -h -t https://github.com/michaelbromley/angularUtils.git.

Able to execute same from shell without a problem...
Attached are the log files.  This is probably an easy fix, but at this time 
not clear to me...
[2018-07-19T14_03_25_500Z-debug.log](https://github.com/openbmc/openbmc/files/2210980/2018-07-19T14_03_25_500Z-debug.log)
[log.do_compile.16677.log](https://github.com/openbmc/openbmc/files/2210989/log.do_compile.16677.log)

gtmills, 2018-08-14:
@timruggles Were you successful in building an image or are you still seeing this fail? 
This looks like a connection issue. 
We are looking at removing https://github.com/michaelbromley/angularUtils.git.

timruggles, 2018-08-16:
I never was able to resolve the issue. I ended up just removing webui from the build. The plan was to re-visit the issue at a later date.  It has something to do with the corp firewall, but ran out of time trying to resolve it.

closed 2018-08-16 by timruggles