openbmc/openbmc #3318: Build failure with perf

alehed, 2018-07-20:
Building perf fails in the do_configure step because a `make clean` step doesn't find the files it should delete.

I assume this is caused by a missing ordering constraint. A temporary workaround is just to build again.

The failure log says:

```
DEBUG: Executing python function copy_perf_source_from_kernel
DEBUG: Python function copy_perf_source_from_kernel finished
DEBUG: Executing shell function do_configure
NOTE: make -C /path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/tools/perf O=/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0 CROSS_COMPILE=arm-openbmc-linux-gnueabi- ARCH=arm CC=arm-openbmc-linux-gnueabi-gcc  -march=armv7-a -marm -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 --sysroot=/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/recipe-sysroot AR=arm-openbmc-linux-gnueabi-ar LD=arm-openbmc-linux-gnueabi-ld --sysroot=/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/recipe-sysroot  EXTRA_CFLAGS=-ldw EXTRA_LDFLAGS= perfexecdir=/usr/libexec NO_GTK2=1 NO_DWARF=1 NO_LIBAUDIT=1 NO_JVMTI=1 NO_LIBNUMA=1 NO_SDT=1 DESTDIR=/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/image prefix=/usr bindir=/usr/bin sharedir=/usr/share sysconfdir=/etc perfexecdir=/usr/libexec/perf-core ETC_PERFCONFIG=../etc sharedir=share mandir=share/man infodir=share/info clean
make: Entering directory '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/tools/perf'
  CLEAN    libtraceevent
  CLEAN    libapi
  CLEAN    config
  CLEAN    libbpf
  CLEAN    python
  CLEAN    libsubcmd
  CLEAN    fixdep
  CLEAN    libsubcmd
find: '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/python_ext_build': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/util/.usage.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/util/.intel-pt.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/util/hist.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/util/intel-pt.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/util/.rwsem.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/util/.mmap.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/util/parse-events-flex.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/util/.memswap.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.plugin_hrtimer.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.plugin_xen.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/builtin-diff.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.nlattr.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/builtin-bench.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/builtin-version.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/plugin_hrtimer-in.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.plugin_scsi.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.plugin_kmem-in.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/builtin-evlist.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.exec-cmd.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.plugin_function-in.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/plugin_sched_switch.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/kbuffer-parse.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/nlattr.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.help.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.pager.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/libtraceevent-in.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/plugin_cfg80211.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/parse-filter.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.str_error_r.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.plugin_kvm.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/plugin_xen.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/builtin-sched.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/builtin-buildid-list.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/builtin-ftrace.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/plugin_jbd2.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/builtin-lock.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.parse-filter.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/perf-in.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.perf-in.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/libperf-in.o': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.builtin-buildid-list.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/.builtin-mem.o.cmd': No such file or directory
find: cannot delete '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/exec-cmd.o': No such file or directory
[... cut ...]
/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/tools/build/Makefile.include:7: recipe for target 'fixdep-clean' failed
make[1]: *** [fixdep-clean] Error 2
ERROR: oe_runmake failed
Makefile:90: recipe for target 'clean' failed
make: *** [clean] Error 2
make: Leaving directory '/path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/perf-1.0/tools/perf'
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_configure (log file is located at /path/to/openbmc/build/tmp/work/evb_zx3_pm3-openbmc-linux-gnueabi/perf/1.0-r9/temp/log.do_configure.3970)

```

mine260309, 2018-07-23:
I hit this issue as well, and workaround it by clean and rebuild.

mine260309, 2018-08-24:
A minimal steps to reproduce the issue with linux kernel itself:
```
# linux kernel source at ~/work/linux
mkdir tempdir
make -C tools/perf O=~/work/linux/tempdir all  # make perf and output to tempdir, it builds ok
make -C tools/perf O=~/work/linux/tempdir clean  # it gets error as this issue
```

mine260309, 2018-08-24:
Added a patch in https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc/+/12101/

@shenki Could you kindly help to check if this patch should be sent to linux kernel upstream?
If yes, we could fix this in kernel and then remove the above commit in openbmc.

rfrandse, 2018-08-24:
<https://gerrit.openbmc-project.xyz/12101> perf: Fix build error on rebuilding
Resolves  #3318 Build failure with perf

mine260309, 2018-09-21:
This should be closed as the fix is already merged.

shenki, 2018-09-21:
This was fixed upstream in https://github.com/torvalds/linux/commit/e04a365aacafe49a034a9da026a502d5c147f3e8. We cherry picked that into dev-4.18 and this was merged into openbmc as of https://github.com/torvalds/linux/commit/e04a365aacafe49a034a9da026a502d5c147f3e8.

closed 2018-09-21 by shenki