openbmc/openbmc #3319: Witherspoon: Services core-dump during sutdown  when BMC is rebooted

gkeishin, 2018-07-20:
```
Jul 20 15:01:39 witherspoon systemd[1]: Stopped Phosphor MBOX Daemon.
Jul 20 15:01:39 witherspoon systemd[1]: witherspoon-fan-watchdog.service: Main process exited, code=killed, status=9/KILL
Jul 20 15:01:39 witherspoon systemd[1]: witherspoon-fan-watchdog.service: Failed with result 'signal'.
Jul 20 15:01:39 witherspoon systemd[1]: Stopped Witherspoon Fan Watchdog Daemon.
Jul 20 15:01:39 witherspoon systemd[1]: Stopped UART render controller.
Jul 20 15:01:39 witherspoon systemd[1]: Stopped Network IPMI daemon.
Jul 20 15:01:39 witherspoon systemd[1]: Stopped Phosphor Time Manager daemon.
Jul 20 15:01:39 witherspoon systemd[1]: Stopped The NGINX HTTP and reverse proxy server.
Jul 20 15:01:39 witherspoon systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 20 15:01:39 witherspoon systemd[1]: systemd-coredump.socket: Failed to queue service startup job (Maybe the service file is missing or not a non-template unit?): Transaction is destructive.
Jul 20 15:01:39 witherspoon systemd[1]: systemd-coredump.socket: Failed with result 'resources'.
Jul 20 15:01:39 witherspoon systemd[1]: xyz.openbmc_project.Inventory.Manager.service: Main process exited, code=dumped, status=6/ABRT
Jul 20 15:01:39 witherspoon systemd[1]: xyz.openbmc_project.Inventory.Manager.service: Failed with result 'core-dump'.
Jul 20 15:01:39 witherspoon systemd[1]: Stopped Phosphor Inventory Manager.
Jul 20 15:01:39 witherspoon systemd[1]: xyz.openbmc_project.Logging.service: Main process exited, code=dumped, status=6/ABRT
Jul 20 15:01:39 witherspoon systemd[1]: xyz.openbmc_project.Logging.service: Failed with result 'core-dump'.
Jul 20 15:01:39 witherspoon systemd[1]: Stopped Phosphor Log Manager.
Jul 20 15:01:39 witherspoon systemd[1]: xyz.openbmc_project.LED.GroupManager.service: Main process exited, code=dumped, status=6/ABRT
Jul 20 15:01:39 witherspoon systemd[1]: xyz.openbmc_project.LED.GroupManager.service: Failed with result 'core-dump'.
Jul 20 15:01:39 witherspoon systemd[1]: Stopped Phosphor LED Group Management Daemon.
Jul 20 15:01:39 witherspoon systemd[1]: xyz.openbmc_project.Software.BMC.Updater.service: Main process exited, code=dumped, status=6/ABRT
Jul 20 15:01:39 witherspoon systemd[1]: xyz.openbmc_project.Software.BMC.Updater.service: Failed with result 'core-dump'.
Jul 20 15:01:39 witherspoon systemd[1]: Stopped OpenBMC Software Update Manager.
Jul 20 15:01:39 witherspoon systemd[1]: xyz.openbmc_project.State.Host.service: Main process exited, code=dumped, status=6/ABRT
Jul 20 15:01:39 witherspoon systemd[1]: xyz.openbmc_project.State.Host.service: Failed with result 'core-dump'.
```

- [x] Journald 
[Journald_BMC_reboott.txt](https://github.com/openbmc/openbmc/files/2214430/Journald_BMC_reboott.txt)

geissonator, 2018-07-20:
It's not clear to me yet why, but the implicit dependency you get when you set a "Type=dbus" in your service on dbus.socket does not get you a dependency on dbus.service.

So in this case we've got dbus.service stopping before service that require it have been stopped. The simple fix is to put a dependency in mapper on the dbus.service.

https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc/+/11568/

gtmills, 2018-07-20:
@geissonator @gkeishin When did this start? Is this a problem we have always had that now is exposed because of the sdbusplus change?

rfrandse, 2018-07-20:
<https://gerrit.openbmc-project.xyz/11568> Add explicit dependency between mapper and sdbus
Resolves  #3319 Witherspoon: Services core-dump during shutdown path when BMC is rebooted

closed 2018-07-20 by geissonator