openbmc/openbmc #3320: Code update fails with: libubi: error!: cannot read "/sys/class/ubi/ubi8_0/type"

anoo1, 2018-07-20:
A Delete version followed by an Activate may fail with error message with the latest master:
```
Jul 16 04:32:00 witherspoon kernel[985]: [[0;1;39m[  160.080000] UBIFS (ubi8:1): un-mount UBI device 8[[0m
Jul 16 04:32:00 witherspoon kernel[985]: [[0;1;39m[  160.080000] UBIFS (ubi8:1): background thread "ubifs_bgt8_1" stops[[0m
Jul 16 04:32:00 witherspoon obmc-flash-bios[1334]: libubi: error!: cannot read "/sys/class/ubi/ubi8_0/type"
Jul 16 04:32:00 witherspoon obmc-flash-bios[1334]:         error 19 (No such device)
Jul 16 04:32:00 witherspoon obmc-flash-bios[1334]: ubirmvol: error!: cannot find UBI volume "pnor-rw-d8368f51"
Jul 16 04:32:00 witherspoon obmc-flash-bios[1334]:           error 19 (No such device)
Jul 16 04:32:00 witherspoon systemd[1]: Started Remove UBIFS volume pnor-rw-d8368f51.
Jul 16 04:32:01 witherspoon systemd[1]: Started Remove UBIFS volume pnor-ro-d8368f51.
Jul 16 04:32:01 witherspoon kernel: block ubiblock8_0: released
Jul 16 04:32:01 witherspoon kernel[985]: [  161.040000] block ubiblock8_0: released
```

anoo1, 2018-07-20:
There was a regression in mtd-utils, causing this error if the deleted version was not the last volume:

http://lists.infradead.org/pipermail/linux-mtd/2018-June/081562.html

rfrandse, 2018-07-20:
<https://gerrit.openbmc-project.xyz/11571> mtd-utils: Revert "Return correct error number in ubi_get_vol_in"
Fixes  #3320 Code update fails with: libubi: error!: cannot read "/sys/class/ubi/ubi8_0/type"

closed 2018-11-12 by rfrandse