openbmc/openbmc #3321: Witherspoon: Front ID LED is ON when host is off

gkeishin, 2018-07-24:
Driver:  Master level 
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.3-68"
VERSION_ID="v2.3-68-gcffe1bf"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.3-68"
BUILD_ID="v2.3"
```

State:
```
/xyz/openbmc_project/state/enumerate
{
    "/xyz/openbmc_project/state/bmc0": {
        "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
        "LastRebootTime": 0,
        "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
    },
    "/xyz/openbmc_project/state/chassis0": {
        "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.Off",
        "POHCounter": 5,
        "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    },
    "/xyz/openbmc_project/state/host0": {
        "AttemptsLeft": 3,
        "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.Unspecified",
        "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
        "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.Inactive",
        "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
    }
}
```

LED:
```
Executing: GET  /xyz/openbmc_project/led/physical/front_id

/xyz/openbmc_project/led/physical/front_id
{
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.On"
}

Executing: GET  /xyz/openbmc_project/led/physical/rear_id

/xyz/openbmc_project/led/physical/rear_id
{
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.Off"
}
```

gkeishin, 2018-07-24:
```
Test Suite File		: Tests.Test Basic Poweron 
Test Case Name		: Verify Front And Rear LED At Standby
Test Source File	: /home/fspcibld/openbmc-test-automation/tests/test_basic_poweron.robot
Failure Time Stamp	: 20180724170610174829
Test Error Message	: Unexpected LED state.: xyz.openbmc_project.Led.Physical.Action.On != xyz.openbmc_project.Led.Physical.Action.Off
Test Documentation	:
Front and Rear LED should be off at standby.
```

- [x] Journald 
[LEDs.txt](https://github.com/openbmc/openbmc/files/2224952/LEDs.txt)

gtmills, 2019-01-23:
@gkeishin Still a problem?

gkeishin, 2019-01-23:
Not anymore since quite time.. Closing now

closed 2019-01-23 by gkeishin