openbmc/openbmc #3322: os-release build failure on Power

mine260309, 2018-07-25:
On latest master (38828d457a832840b186a1e27c08bf9ae2284ee0), `os-release.bb` gets build error as below:
```
Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "ppc64le-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "romulus"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm armv6 thumb arm1176jzs"
TARGET_FPU           = "soft"
meta
meta-poky
meta-oe
meta-networking
meta-perl
meta-python
meta-webserver
meta-virtualization
meta-phosphor
meta-aspeed
meta-openpower
meta-ibm
meta-romulus         = "master:38828d457a832840b186a1e27c08bf9ae2284ee0"

Initialising tasks: 100% |############################################################################################################################################| Time: 0:00:00
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: os-release-1.0-r0 do_package_write_rpm: Function failed: BUILDSPEC (log file is located at /home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/temp/
log.do_package_write_rpm.115200)
ERROR: Logfile of failure stored in: /home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/temp/log.do_package_write_rpm.115200
Log data follows:
| DEBUG: Executing python function sstate_task_prefunc
| DEBUG: Python function sstate_task_prefunc finished
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/leiyu/work/openbmc-2/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/leiyu/work/openbmc
-2/meta/recipes-devtools/rpm/rpm_4.14.1.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: ['rpm-native', 'gnu-config-native', 'python3-native', 'bzip2-native', 'autoconf-native', 'automake-native', 'dbus-native', 'file-native', 'libarchive
-native', 'popt-native', 'db-native', 'libtool-native', 'gettext-minimal-native', 'nss-native', 'quilt-native', 'pkgconfig-native', 'elfutils-native', 'xz-native', 'sqlite3-native', 'gdbm-native', 'readline-native', 'openssl-native', 'zlib-native', 'm4-native', 'texinfo-dummy-native', 'expat-native', 'autoconf-archive-native', 'lzo-native', 'e2fsprogs-native',
 'nspr-native', 'ncurses-native', 'makedepend-native', 'cryptodev-linux-native', 'util-linux-native', 'attr-native', 'util-macros-native', 'xproto-native']
| NOTE: Skipping as already exists in sysroot: ['pseudo-native']
| DEBUG: sed -e 's:^[^/]*/:/home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/recipe-sysroot-native/:g' /home/leiyu/work/openbmc-2/build/tmp/sysroots-comp
onents/ppc64le/rpm-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-components/ppc64le/gnu-config-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-compon
ents/ppc64le/python3-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-components/ppc64le/autoconf-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-compon
ents/ppc64le/automake-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-components/ppc64le/dbus-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-component
s/ppc64le/libtool-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-components/ppc64le/quilt-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-components/p
pc64le/pkgconfig-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-components/ppc64le/elfutils-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-components
/ppc64le/openssl-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-components/ppc64le/e2fsprogs-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-component
s/ppc64le/nspr-native/fixmepath /home/leiyu/work/openbmc-2/build/tmp/sysroots-components/ppc64le/ncurses-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/home/leiyu/work
/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/r
ecipe-sysroot-native:g' -e 's:FIXME_COMPONENTS_DIR:/home/leiyu/work/openbmc-2/build/tmp/sysroots-components:g' -e 's:FIXME_HOSTTOOLS_DIR:/home/leiyu/work/openbmc-2/build/tmp/hosttoo
ls:g' -e 's:FIXME_PKGDATA_DIR:/home/leiyu/work/openbmc-2/build/tmp/pkgdata/romulus:g' -e 's:FIXME_PSEUDO_LOCALSTATEDIR:/home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os
-release/1.0-r0/pseudo/:g' -e 's:FIXME_LOGFIFO:/home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/temp/fifo.115200:g'
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing python function do_package_write_rpm
| DEBUG: Executing python function read_subpackage_metadata
| DEBUG: Python function read_subpackage_metadata finished
| DEBUG: Executing python function do_package_rpm
| DEBUG: Executing python function write_specfile
| NOTE: Creating EMPTY RPM Package for os-release-dbg
| NOTE: Not creating empty RPM package for os-release-staticdev
| NOTE: Creating EMPTY RPM Package for os-release-dev
| NOTE: Not creating empty RPM package for os-release-doc
| NOTE: Not creating empty RPM package for os-release-locale
| NOTE: Creating RPM package for os-release
| NOTE: Creating RPM package for os-release
| DEBUG: Python function write_specfile finished
| DEBUG: PKGWRITEDIR: /home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/deploy-rpms/noarch
| DEBUG: Executing shell function BUILDSPEC
| Building target platforms: noarch-openbmc-linux
| Building for target noarch-openbmc-linux
| Processing files: os-release-1.0-r0.noarch
| warning: Deprecated external dependency generator is used!
| Finding  Provides:
| Finding  Requires(interp):
| Finding  Requires(rpmlib):
| Finding  Requires(verify):
| Finding  Requires(pre):
| Finding  Requires(post):
| Finding  Requires(preun):
| Finding  Requires(postun):
| Finding  Requires(pretrans):
| Finding  Requires(posttrans):
| Finding  Requires:
| Finding  Conflicts:
| Finding  Obsoletes:
| Finding  Recommends:
| Finding  Suggests:
| Finding  Supplements:
| Finding  Enhances:
| Provides: os-release = 1.0-r0
| Requires(rpmlib): rpmlib(CompressedFileNames) <= 3.0.4-1 rpmlib(FileDigests) <= 4.6.0-1 rpmlib(PayloadFilesHavePrefix) <= 4.0-1
| Processing files: os-release-dbg-1.0-r0.noarch
| Processing files: os-release-dev-1.0-r0.noarch
| Checking for unpackaged file(s): /home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/recipe-sysroot-native/usr/bin/../../usr/lib/rpm/check-files /home/le
iyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/package
| Wrote: /home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/deploy-rpms/noarch/os-release-1.0-r0.noarch.rpm
| Finished binary package job, result 0, filename /home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/deploy-rpms/noarch/os-release-1.0-r0.noarch.rpm
| Segmentation fault (core dumped)
| WARNING: exit code 139 from a shell command.
| DEBUG: Python function do_package_rpm finished
| DEBUG: Python function do_package_write_rpm finished
| ERROR: Function failed: BUILDSPEC (log file is located at /home/leiyu/work/openbmc-2/build/tmp/work/all-openbmc-linux/os-release/1.0-r0/temp/log.do_package_write_rpm.115200)
ERROR: Task (/home/leiyu/work/openbmc-2/meta/recipes-core/os-release/os-release.bb:do_package_write_rpm) failed with exit code '1'
```

This issue does not occur on x86 machine.

shenki, 2018-07-25:
Which tool is segfaulting? You should be able to check dmesg for the process.

I just did a build on ppc64le, Ubuntu 18.10 (development release), and it succeeded with this patch applied:

 https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/11641

mine260309, 2018-07-25:
It is `rpmbuild.real` that gets segment fault.
I will make a fresh clone and see if it's due to bad sstate.

mine260309, 2018-07-26:
Did a fresh build without `sstate-cache`, and it passes.
Close it.

closed 2018-07-26 by mine260309