openbmc/openbmc #3327: IPMI Chassis identify 0 is not getting identify LED to go in off state.

rahulmah, 2018-07-26:
IPMI Chassis identify 0 is not getting identify LED to go in off state. I waited for almost 15 mins.

```
bash-4.1$ ipmitool -I lanplus -C 3 -P 0penBmc -H <BMC_IP>.aus.stglabs.ibm.com chassis identify force
Chassis identify interval: indefinite
bash-4.1$ curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/led/physical/front_id
{
  "data": {
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.Blink"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ipmitool -I lanplus -C 3 -P 0penBmc -H <BMC_IP>.aus.stglabs.ibm.com chassis identify 0
Chassis identify interval: off
bash-4.1$ curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/led/physical/front_id
{
  "data": {
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.Blink"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/led/physical/front_id
{
  "data": {
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.Blink"
  },
  "message": "200 OK",
  "status": "ok"
```

Fw info:
BMC: v2.3-93-g71b5d58
PNOR: IBM-witherspoon-ibm-OP9-v2.0.3-2.17

spinler, 2018-07-26:
@rahulmah did you check the rear identify LED?  Did it behave?

spinler, 2018-07-26:
I can't find any support in the IPMI code for a command to turn off the identify LEDs.  It seems like it just supports it doing it via the timer.  @tomjoseph83 is that right?

rahulmah, 2018-07-27:
@spinler : Yes I check. Same problem was shown by rear identify LED also.

rfrandse, 2018-07-31:
<https://gerrit.openbmc-project.xyz/11711> Handle the case to turn off the Chassis Identify LED
Resolves  #3327 IPMI Chassis identify 0 is not getting identify LED to go in off state.

closed 2018-08-08 by geissonator