openbmc/openbmc #3329: Design and implement OpenBMC REST API to upload LDAP server information

dkodihal, 2018-07-27:
Design an OpenBMC REST API (to be implemented in phosphor-rest-server) to let a user inform the BMC about an external LDAP server configuration.

We've discussed about two approaches :
1) Let the user upload a standard ldap config file. No D-Bus API.
2) A D-Bus API, with properties mapped to specific LDAP config fields.

We'll probably do both, but this story is tracking 1).

What LDAP fields should be provided - URI, port, search base, secure/unsecure. Do you think we need more @ratagupt or @rthomaiy.

The structure of the LDAP config file might depend on the LDAP stack on the BMC (for eg just openldap, or openldap + sssd).

The config upload API could be along the lines of the image upload API. The rest server will have to copy the ldap config to an appropriate location, and restart relevant services. Whether the ret server should delegate all of this to a different application needs more thought.

This LDAP provisioning would require an existing local user on the BMC (most likely root).

@bradbishop, @edtanous do you have thoughts on this?

edtanous, 2018-07-30:
I suspect we need a greater discussion around LDAP support in general.  While this task seems like a great feature to have, it's going to be driven by whatever design is pushed upstream by the teams that are working on it.  Some key things that are going to be needed before this task can even be scoped are:
1. Appropriate implementation of user management in OpenBMC.  This work is still being driven by several people on several teams, and is in a bit of flux.  Without appropriate user management support (create/rename/change password/change permissions) I'm not sure how anyone would stand a chance of implementing LDAP.
2. A design document about how LDAP is configured and architected.  Do users authenticate as themselves on the BMC, or are all LDAP logins coalesced into a single user account?  How are permissions and roles handled, by a single query, or multiple queries?
 
You mention the openldap or sssd config files.  Whatever API we write should be backend agnostic.  It seems very likely that people may want to change out the backends, if for example, they don't have room for sssd in flash.  We should support that, and implement a _generic_ API that could be implemented by any one of a number of backends.

In a perfect world, we could also anticipate the addition of ActiveDirectory support, but I don't have very much knowledge there, and my  products don't support it.

Ideally this API would be implemented as a Redfish function.  There are already several examples of Redfish servers that implement LDAP config support, so it's very much possible, just needs someone with the willpower to get a patchset up.

ratagupt, 2018-08-01:
> Appropriate implementation of user management in OpenBMC. This work is still being driven by several people on several teams, and is in a bit of flux. Without appropriate user management support [...]

ratagupt, 2018-08-09:
@dkodihal  @ngorugan  Can we use the story https://github.com/openbmc/openbmc/issues/3305 as this story was to track the option 1)(Let the user upload a standard ldap config file. No D-Bus API.) which we are not doing now?

ratagupt, 2019-04-19:
Can be closed now.

closed 2019-10-21 by rfrandse