openbmc/openbmc #3330: Romulus: get multiple checkstops after inject error

mine260309, 2018-07-27:
While doing error injection test on Romulus, an issue is found that it gets multiple checkstops after BMC reboots the host.

Typically, after error inject, the expected behavior is:
1. BMC waits for 10s, and reboot host;
2. Host boots to ISTEP6.9, gard the error parts, send eSELs, and ask for reboot;
3. BMC reboots host again;
4. Host boots OK.

When the issue occurs, the behavior becomes:
1. BMC waits for 10s, and reboot host;
2. Host boots and BMC gets checkstop again;
3. Loop of 1-2 for some times;
   3.1 Sometimes, host boots to ISTEP6.9, gard the error parts, send eSELs, and ask for reboot, then it's OK.
   3.2 Sometimes, it continue the loop of 1-2, and eventually BMC does not reboot host anymore, and keep host in Quiesce state.

The hostboot log when issues occurs:
```
--== Welcome to Hostboot  ==--

  2.85924|secure|SecureROM valid - enabling functionality
  6.02400|IPMI: shutdown requested

--== Welcome to Hostboot  ==--

  2.85932|secure|SecureROM valid - enabling functionality
  6.02629|IPMI: shutdown requested

--== Welcome to Hostboot  ==--

  2.85934|secure|SecureROM valid - enabling functionality
  6.02491|IPMI: shutdown requested
```

Tested this case on both Witherspoon and Romulus, and it only occurs on Romulus.

mine260309, 2018-07-27:
In `phosphor-reboot-host@.service`, it sleeps for 5 seconds between power is shutdown and reboot the host.
If I change this value to 8 seconds, the issue does not occur anymore.

So I guess this seems to indicate that the CPU error status is not cleared during reboot.

kyle-ibm, 2018-07-31:
verified on my romulus that increasing sleep to 8 sec avoids the timing issue.

Though we should still probably look into why there is a timing issue on clearing the checkstop signal. 
Noting here that Witherspoon does not see this issue.

mine260309, 2018-08-01:
https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc/+/11716/

rfrandse, 2018-08-01:
<https://gerrit.openbmc-project.xyz/11716> meta-romulus: Increase delay in host reboot
Resolves  #3330 Romulus: get multiple checkstops after inject error

closed 2018-08-08 by geissonator