openbmc/openbmc #3331: Design and implement OpenBMC REST API to upload SSL certificate

dkodihal, 2018-07-27:
We've discusses couple of options :

1) Asking the user to upload to the BMC a signed certificate plus private key.
2) Having the BMC generate a Certificate Signing Request and private key. User downloads CSR, gets it signed, and then uploads.

2) is deemed more secure (although more work), and this story tracks 2).

Would it be even technically possible to support 1) and 2) on a BMC?

The API itself has to be an OpenBMC REST API. Nothing on the D-Bus. The phosphor rest-server may actually delegate the task of copying the uploaded certificate, generating the CSR, restarting services, etc to another application. There would need to be a CSR download API, and a certificate upload API.

The implementation would have to do CSR generation, validation of uploaded certificate (is this needed?), copy the certificate to a relevant location, and restart specific services (for eg nginx).

A user may want to user different certificates for a server/client based provisioning. So for example the BMC could be a web server and an LDAP client. Different certificates are preferred in this case, so potentially different REST APIs, or maybe the API says what's the target service of the certificate.

Thoughts @ratagupt, @bradbishop and @edtanous?

edtanous, 2018-07-30:
All the BMCs I've been a part of have implemented #1, and allowed uploading a private key and a cert, sometimes as a single file, sometimes as separate files.  #1 is verymuch technically possible, as it's already implemented in practice.
#2, while supposedly more secure, leaves a number of challenges for organizations and dev-ops departments.  I've never heard a request for CSR production from the BMC that wasn't already met by implementing key and cert upload.

For CSR generation, there would be a lot more to implement than I think you've documented, as the APIs would need a way to upload all the different fields needed for the CSR.

For what it's worth, I would recommend implementing this new API as a Redfish OEM schema, under the managers root object.  Redfish has a nice way (through schema files) that we can manage and introspect the REST api itself, and there are already several examples of shipping systems that implement redfish APIs, that proves that it can be done in production.

"validation of uploaded certificate (is this needed?)"'
Yes, although it should only be verified for content, not necessarily the whole cert chain, as we might not have access to the full chain.

"A user may want to user different certificates for a server/client based provisioning"
I'm not really following this statement.  Up until this point I assumed you were talking about TLS hosting certificates.  In the cases where the bmc is the client, that interacts with the root certificate store, and is a bit more complicated of an issue, and one that we likely should push to not solve until we absolutely need it.

joseph-reynolds, 2018-08-02:
Straw-man proposal for SSL certificate REST APIs
I had posted this on the email list (lightly edited here):

URIs:

- /xyz/openbmc_project/certificates/ssl/{index}
  - Represents a SSL certificate
  - Supports GET, PUT, POST, DELETE
  - JSON { "certificate": "..." }
- /xyz/openbmc_project/certificates/ssl/active
  - Represents which certificate is used by SSL
  - Supports GET, PUT, DELETE
  - JSON { "data": "/xyz/openbmc_project/certificates/ssl/1" }
  - See the magic below.

Notes:
 - These represent a collection of SSL certificates and optionally an active one.
 - The number of certificates can be zero or more.
 - The certificate numbering ({index}) must be stable across API operations.
 - The implementation can impose a maximum limit (16 is plenty?).
 - Duplicate certificates are allowed and are not detected.
 - Expired certificates are allowed and are not detected.
 - Mal-formed certificates are detected? TODO
 - The certificates and active certificate info persists across rebooting.

Magic for the active SSL certificate:
 - When SSL is started with no active certificate:
   + A certificate is created and used to sign itself.
   + The new certificate becomes the active certificate.
   + TODO: What if you already have the maximum number of certificates allowed?
 - When you attempt to delete the active certificate:
   + This is allowed.  TODO: Or maybe not allowed.
   + There will be no active certificate (see implementation note).
   + /xyz/openbmc_project/certificates/ssl/active is deleted.
   + The underlying certificate is not deleted.
 - When you PUT (replace) the active certificate:
   + This is allowed.  TODO: Or maybe not allowed.
   + The newly put certificate becomes active (see implementation note).
   + /xyz/openbmc_project/certificates/ssl/active does not change.

Implementation note: When you change the active certificate while SSL is running, does the behavior of SSL change immediately, or only for new connections?? NOTE: There was discussion in the email list about this after I write this question.

Ideas to support future behaviors:

1. To support future certificate types, the ssl path element was used. Ideas:
  /xyz/openbmc_project/certificates/ssl
  /xyz/openbmc_project/certificates/ldap
  /xyz/openbmc_project/certificates/host
  /xyz/openbmc_project/certificates/rootoftrust

2. There are two options to articulate the content of each certificate (content such as an associated private key, expiration, or certificate signing request): JSON and URIs:
JSON: { "certificate": "...", "privatekey": "...", "expiration": "DATE" }
URIs:
  /xyz/openbmc_project/certificates/ssl/1/certificate
  /xyz/openbmc_project/certificates/ssl/1/privatekey
  /xyz/openbmc_project/certificates/ssl/1/csr
The JSON approach seems way easier.

3. The URIs for certificate signing requests (CSR) could be:
  /xyz/openbmc_project/certificates/csr/{index}
  /xyz/openbmc_project/certificates/ssl/csr
  /xyz/openbmc_project/certificates/ssl/csr/{index}
depending on how many different CSRs we need to support.

dkodihal, 2018-11-07:
REST API has been merged.

closed 2018-11-07 by dkodihal