openbmc/openbmc #3334: Why can‘t open the WEB UI？?

FrankHome, 2018-08-02:
1.ip can ping success
2.ssh can login success
3.ipmi command can send success
4.had build phosphor-webui
5.bmcweb.service is runing 
But Always can not open the web ui 
Need your help, Thanks a lot!!
[image]
[image]
[image]

sivassrr, 2018-08-02:
Have you tried with https? and also you may want to accept the advance option "proceed to <ip>". (case of chrome browser). Screen will be as shown below. 
[image]

FrankRenn, 2018-08-02:
@sivassrr 
Hi sivassrr ,thanks for your help ,but I tried with https, still fail
And I have a question, if I need to install nodejs in obmc system ?
[image]
[image]

sivassrr, 2018-08-02:
With the official image I see  in systemctl output
```
  nginx.service                              loaded    active   running   The NGINX HTTP and reverse```

As you are running with your sandbox, Pls check whether it is running.

FrankRenn, 2018-08-02:
After check ,I find there is no nginx.service in obmc system, so i try to add and build ,but occur the fellow error:
[image]
The build system is ubuntu14.04
Any suggestion about that?

FrankRenn, 2018-08-03:
The issue has solved ,so close the issue

closed 2018-08-03 by FrankRenn

sivassrr, 2018-08-03:
Good to hear, Issue is resolved. Could you pls put how it got resolved that helps. If any one checks the history of this issue.

michealylj1, 2019-06-03:
Hi FrankRenn,

> At first can not login webui because have no bmcweb.socket in obmc system.
> The nodejs-native do_compile fail because my npm have some problem
> At last to login web ui again,need to appoint port number883: https://10.0.15.177:883
> Used the meta-x86/meta-intel/meta-s2600wf

I just meet the same issue with you when I run openbmc in the chip "meta-evb-ast2500", Please tell me 

1) how to add nginx service in obmc system?
2) how to enable bmcweb.socket  in obmc system?
3) If I need to install nodejs in obmc system, how to do ?

Thanks,
Michael