openbmc/openbmc #3335: Romulus: add missing ipmi configs

mine260309, 2018-08-02:
There are quite a lot of ipmi related configs (yaml, json, etc) are needed.
On Witherspoon, most of the yaml configs are generated from MRW, and there are some hard-coded configs as well.

On Romulus, there are only hard-coded configs, and it is missing some configs, e.g.
* Temperatures of cpu, dimm;
* Entity id and entity instances;
* sensorNamePattern of each sensor

The goal of this issue is to follow Witherspoon's generated configs, and add the missing pieces for Romulus.

mine260309, 2018-08-03:
Added https://github.com/mine260309/openbmc-tools/blob/dev-mine/leiyu/obmc-utils/sensor_yaml_config.py which will fix the `entityID` `entityInstance`.

The script should be improved to generate the whole sensor config from rpt file.

tomjoseph83, 2018-08-07:
1) phosphor-ipmi-sensor-inventory: For sensorNamePattern, the ideal thing will be look at the generated yaml for Witherspoon. recipes-phosphor/ipmi/phosphor-ipmi-sensor-inventory-mrw-config/config.yaml is the configuration and operates on the MRW output.

2) phosphor-ipmi-inventory-sel: This yaml is needed to map the D-Bus error log object to the SEL entry. If there is FRU callout, it will be shown as an appropriate SEL entry

3) phosphor-ipmi-fru-properties: This yaml is to map FRU ID to Inventory D-Bus objects.

4) Conf files for IPMI: There is bunch of IPMI/DCMI specific configuration files in meta-witherspoon/recipes-phosphor/ipmi/witherspoon-ipmi-config/, something similar is needed for meta-romulus.

5) meta-witherspoon/recipes-phosphor/ipmi/phosphor-ipmi-fru-read-bmc-inventory% contains the FRU ID mapping for BMC FRU's. This is needed for ipmitool fru print to show these FRU's as well.

6) meta-witherspoon/recipes-phosphor/ipmi/phosphor-ipmi-fru-read-not-sent-by-host-inventory% - This mapping is needed for the GPU's.

mine260309, 2018-08-08:
Thanks @tomjoseph83 

For Romulus:

* phosphor-ipmi-sensor-inventory
   * `sensorNamePattern` is to be added
   * core/dimm temp sensors are added in https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc/+/11838/

* phosphor-ipmi-inventory-sel
   This is already done in 243a674

* phosphor-ipmi-fru-properties
   This is already done in 588b9f8, but the `Cached` property is true, while in Witherspoon it is `false`. Should this be fixed?

* Conf files for IPMI:
   To be implemented.

* phosphor-ipmi-fru-read-bmc-inventory%
   Submitted https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc/+/11751/

* phosphor-ipmi-fru-read-not-sent-by-host-inventory%
   Romulus has no GPUs so it is OK without this.

closed 2019-03-28 by mine260309