openbmc/openbmc #3345: IPMI issues on Palmetto

shenki, 2018-08-03:
Is this expected?
```
Aug 03 05:59:06 palm4-bmc ipmid[920]: getHostNetworkData failed for get_sys_boot_options.
Aug 03 05:59:06 palm4-bmc ipmid[920]: getHostNetworkData failed for get_sys_boot_options.
Aug 03 05:59:06 palm4-bmc ipmid[920]: getHostNetworkData failed for get_sys_boot_options.
Aug 03 05:59:06 palm4-bmc ipmid[920]: getHostNetworkData failed for get_sys_boot_options.
Aug 03 05:59:07 palm4-bmc ipmid[920]: No Registered handlers for NetFn
```

Petitboot logs:
```
/ # cat /var/log/petitboot/pb-discover.log  |grep -i ipmi
  IPMI boot device 0x00
```
```
/ # ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: enP1p6s0f0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether a0:42:3f:20:ca:1e brd ff:ff:ff:ff:ff:ff
    inet 10.61.161.31/16 brd 10.61.255.255 scope global enP1p6s0f0
       valid_lft forever preferred_lft forever
3: enP1p6s0f1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq qlen 1000
    link/ether a0:42:3f:20:ca:1f brd ff:ff:ff:ff:ff:ff
4: tunl0@NONE: <NOARP> mtu 1480 qdisc noop qlen 1000
    link/ipip 0.0.0.0 brd 0.0.0.0
```

How about this? (with host running)
```
$ curl -b /tmp/_xcat_cjar.palm4 -k https://palm4-bmc/xyz/openbmc_project/network/host0/enumerate
{
  "data": {
    "/xyz/openbmc_project/network/host0/intf": {
      "MACAddress": "00:00:00:00:00:00"
    }, 
    "/xyz/openbmc_project/network/host0/intf/addr": {
      "Address": "0.0.0.0", 
      "Gateway": "0.0.0.0", 
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static", 
      "PrefixLength": 0, 
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

mine260309, 2018-08-03:
The settings in `/xyz/openbmc_project/network/host0` are default values.

From IPMI code https://github.com/openbmc/phosphor-host-ipmid/blob/master/chassishandler.cpp#L356-L374:
* if the `MACAddress` is `00:00:00:00:00:00`, it returns -1;
* if the `Address` or `Gateway` is `0.0.0.0`, it returns -1

And it causes the log `getHostNetworkData failed for get_sys_boot_options.`

The settings of `/xyz/openbmc_project/network/host0` is set by IPMI `sys_boot_options` command with `OPAL_NETWORK_SETTINGS` parameter, if it's not set, BMC uses the default ones.

So I would assume this is expected behavior, or Petiboot should set the host interface settings via IPMI.

geissonator, 2018-08-22:
I see ```Aug 03 05:59:06 palm4-bmc ipmid[920]: getHostNetworkData failed for get_sys_boot_options.``` on every boot with withespoon.  @tomjoseph83 to you remember the history on that one? Have we talked with the host team about it?

sammj, 2018-08-23:
It is not an error for the network override settings to not be set (aka at the default values above). Much like bootdev overrides these are intended as a temporary override, not something we expect to always be set.
Given that, constantly logging `failed for get_sys_boot_options` is unnecessary unless a real error has occured.

(Technically it would be more correct for the BMC to return an empty response with the valid flag unset, but that's a bit more work than just dropping an error message.)

closed 2019-04-26 by stale[bot]