openbmc/openbmc #3349: nodejs compilation issue

fengyd2018, 2018-08-07:
source setup  witherspoon
bitbake obmc-phosphor-image

| tls_wrap.cc:(.text+0x31fd): undefined reference to `BIO_ctrl'
| tls_wrap.cc:(.text+0x32f7): undefined reference to `ERR_set_mark'
| tls_wrap.cc:(.text+0x331b): undefined reference to `SSL_write'
| tls_wrap.cc:(.text+0x335a): undefined reference to `ERR_pop_to_mark'
| /share/witherspoon/openbmc/build/tmp/work/x86_64-linux/nodejs-native/8.9.4-r0/node-v8.9.4/out/Release/obj.target/node/src/tls_wrap.o: In function `node::TLSWrap::TLSWrap(node::Environment*, node::crypto::SSLWrap<node::TLSWrap>::Kind, node::StreamBase*, node::crypto::SecureContext*)':
| tls_wrap.cc:(.text+0x36e7): undefined reference to `SSL_new'
| tls_wrap.cc:(.text+0x3901): undefined reference to `SSL_CTX_sess_set_get_cb'
| tls_wrap.cc:(.text+0x3916): undefined reference to `SSL_CTX_sess_set_new_cb'
| /share/witherspoon/openbmc/build/tmp/work/x86_64-linux/nodejs-native/8.9.4-r0/node-v8.9.4/out/Release/obj.target/node/src/tls_wrap.o: In function `node::crypto::SSLWrap<node::TLSWrap>::~SSLWrap()':

Summary: 1 task failed:
  virtual:native:/share/witherspoon/openbmc/import-layers/meta-openembedded/meta-oe/recipes-devtools/nodejs/nodejs_8.9.4.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

gtmills, 2018-08-29:
@fengyd2018 Still seeing this?

HankChang, 2018-08-30:
It seems like the same issue here #3289 
(which could be solved by adjusting the link order in its cctest.target.mk, putting -lcrypto and -lssl at last; or upgrading ld to v2.26)

closed 2019-04-26 by stale[bot]