openbmc/openbmc #3352: phosphor-dbus-monitor: elog doesn't work for deferrable callbacks

AlexanderAmelkin, 2018-08-08:
As we're trying to implement extended temperature-based shutdown we've stumbled at this behavior in pdm:

1. Method `watch->callback(Context::START)` is called on program start from `main()` function in main.cpp in a loop: `for (auto& watch : ConfigPropertyWatches::get())`;
2. If there was any condition in yaml file with `defer` attribute, a `DeferrableCallback` is created for it. `operator()` is called then for `DeferrableCallback` on program start with `Context::START` as an argument;
3. Object `timer` is created on the first `operator()` call in `DeferrableCallback`. The argument to `operator()`, which is `Context::START`, is passed to the timer constructor and is saved there;
4. Timer's context is not changed on subsequent calls to `operator()`;
5. When the `timer` expires, it calls `operator()` with the saved `Context::START` context, always;
6. If the next callback is a DBus method call, the context is not checked and the method is called, but if the next callback is `ElogWithMetadataCapture`, the context is checked and the method is not called for `Context::START`

Due to that, we now can not log sensor threshold crossing if we want to do it with a delay and an extra check for persistence of the condition.

We'd like to know if it's a bug or an intended behavior? Maybe we must use some different approach to logging/acting on threshold crossing?

mine260309, 2018-08-08:
@spinler may have a look into this.

spinler, 2018-08-09:
@AlexanderAmelkin That would be good to get fixed.  Do you have a proposal?

geissonator, 2018-08-22:
Issue moved to [openbmc/phosphor-dbus-monitor #1](https://github.com/openbmc/phosphor-dbus-monitor/issues/1) via [**ZenHub**](https://www.zenhub.com/)

closed 2018-08-22 by geissonator