openbmc/openbmc #3353: Witherspoon: HTTP response header doesnt comply with Redfish Spec

gkeishin, 2018-08-08:
```
$ curl --insecure -X POST -D headers.txt https://xx.xx.xx.xx/redfish/v1/SessionService/Sessions -d '{"UserName":"root", "Password":"0penBmc"}'
{
  "@odata.context": "/redfish/v1/$metadata#Session.Session",
  "@odata.id": "/redfish/v1/SessionService/Sessions/H6aqsDdd0c",
  "@odata.type": "#Session.v1_0_2.Session",
  "Description": "Manager User Session",
  "Id": "H6aqsDdd0c",
  "Name": "User Session",
  "UserName": "root"
}
```

```
$ cat headers.txt
HTTP/1.1 200 OK
Server: nginx
Date: Wed, 08 Aug 2018 12:33:51 GMT
Content-Type: application/json
Content-Length: 283
Connection: keep-alive
Keep-Alive: timeout=5
X-Auth-Token: 36QnhdXttmImzekgq3Wh
Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
X-UA-Compatible: IE=11
X-Frame-Options: DENY
X-XSS-Protection: 1; mode=block
X-Content-Security-Policy: default-src 'self'
Pragma: no-cache
Cache-Control: no-Store,no-Cache
```

From the above header data, the `Location` is not part of the header. 
As per http://redfish.dmtf.org/schemas/DSP0266_1.3.0.pdf  spec
[image]

So DMTF API library requires this `Location` data is a must to be able to clean delete the new session resource we create when requested once we are done.

Any reason why we haven't send the `Location`  in the header here in BMC.  To me its a bug here.

geissonator, 2018-08-08:
@edtanous is this the issue we were discussing in IRC yesterday where I don't have bmcweb serving up the redfish schemas?

edtanous, 2018-08-13:
Yes, this is a bug, fixed in the review below, and isn't specific to witherspoon.  I've added both of you to the review.  It would be good if you could comment on if it fixes the problem for you guys.
https://gerrit.openbmc-project.xyz/#/c/openbmc/bmcweb/+/11723/

gkeishin, 2018-08-13:
Thanks, @edtanous

gtmills, 2018-08-14:
https://gerrit.openbmc-project.xyz/#/c/openbmc/bmcweb/+/11723/ was merged. @gkeishin Did it fix the issue? Can this be closed?

closed 2018-08-20 by gkeishin