openbmc/openbmc #3355: How to access the services provided by  bmcweb

fengyd2018, 2018-08-10:
In the function main of the file src\webserver_main.cpp:
#ifdef BMCWEB_ENABLE_PHOSPHOR_WEBUI
  crow::webassets::requestRoutes(app);
#endif

#ifdef BMCWEB_ENABLE_KVM
  crow::kvm::requestRoutes(app);
#endif

#ifdef BMCWEB_ENABLE_REDFISH
  crow::redfish::requestRoutes(app);
#endif

#ifdef BMCWEB_ENABLE_DBUS_REST
  crow::dbus_monitor::requestRoutes(app);
  crow::image_upload::requestRoutes(app);
  crow::openbmc_mapper::requestRoutes(app);
#endif

It means bmcweb supports webui, kvm, redfish and dbus_rest?
Could someone tell me how to access these services?

gtmills, 2018-08-14:
@edtanous Might be able to answer.

gtmills, 2018-08-14:
@fengyd2018 , this is a good question for the mailing list, as it's not a bug or feature request. You can sign yourself up at https://lists.ozlabs.org/listinfo/openbmc. Thanks!

edtanous, 2018-08-14:
What have you tried so far?  Are you attempting something that hasn't worked?  Have you read the readme in bmcweb?

fengyd2018, 2018-08-20:
Does KVM functionality work? 
How can I interact with the KVM?

edtanous, 2018-09-05:
No.  The KVM is not yet functional.  It is still being developed.

gtmills, 2018-10-12:
Closing since not a bug or feature request. Please reopen if still an issue. Please use IRC or the mailing list for questions. Thanks!!

closed 2018-10-12 by gtmills