openbmc/openbmc #3358: p8: failing to read flash at boot

shenki, 2018-08-14:
I put a 4.18 based kernel (openbmc + 4.18) on the problematic palmetto
we have in Canberra and it started misbehaving even worse than before.
It just reads back FFs.

If I do a `pflash --direct` (ie, use `/dev/mem` to access the controller)
it comes back, but stops working again if I try to power up the host.

I wonder if it's due to running the flash too fast? Here is the output
of pflash doing the tune operation:
```
# pflash --direct -t -g
Flash and controller tuned
AST: CLKIN 24/48Mhz
AST: CPU frequency: 384 Mhz
AST: AHB frequency: 192 Mhz
LIBFLASH: Init status: 02
LIBFLASH: Flash ID: c2.20.1a (c2201a)
AST: Setting up Macronix...
AST: Macronix SR:CR: 0x02:a7
AST: Macronix SR:CR: 0x02:a7
AST: Trying HCLK/5...
  * [00080000] 0 HCLK delay, 4ns DI delay : PASS
  * [00000000] 0 HCLK delay, 0ns DI delay : PASS
  * [00090000] 1 HCLK delay, 4ns DI delay : PASS
AST:  * -> good is pass 1 [0x00000000]
AST: Trying HCLK/4...
  * [00008000] 0 HCLK delay, 4ns DI delay : PASS
  * [00000000] 0 HCLK delay, 0ns DI delay : PASS
  * [00009000] 1 HCLK delay, 4ns DI delay : PASS
AST:  * -> good is pass 1 [0x00000000]
AST: Trying HCLK/3...
  * [00000800] 0 HCLK delay, 4ns DI delay : FAIL
  * [00000000] 0 HCLK delay, 0ns DI delay : PASS
  * [00000900] 1 HCLK delay, 4ns DI delay : PASS
  * [00000100] 1 HCLK delay, 0ns DI delay : PASS
AST:  * -> good is pass 2 [0x00000900]
AST: Trying HCLK/2...
  * [00000980] 0 HCLK delay, 4ns DI delay : FAIL
  * [00000900] 0 HCLK delay, 0ns DI delay : PASS
  * [00000990] 1 HCLK delay, 4ns DI delay : PASS
  * [00000910] 1 HCLK delay, 0ns DI delay : PASS
AST:  * -> good is pass 2 [0x00000990]
AST: Found good read timings at HCLK/2
AST: Command timing set to HCLK/4
LIBFLASH: Found chip Macronix MXxxL51235F size 64M erase granule: 4K
LIBFLASH: Flash >16MB, enabling 4B mode...
LIBFLASH: Enabling controller 4B mode...
```
I can also reproduce this on palm8, which I thought was a production
level machine.

shenki, 2018-08-14:
from @legoater:

> I put a 4.18 based kernel (openbmc + 4.18) on the problematic palmetto
> we have in Canberra and it started misbehaving even worse than before.
> It just reads back FFs.

mine boots fine :/ See the Linux logs below.

are you using the optimize read patchset ? I think so. They should cap 
the speed to 50MHz, but pflash and skiboot are adding some more settings.  

> If I do a pflash --direct (ie, use /dev/mem to access the controller)

what is the 0x1E630004 register setting before

> it comes back, 

and after ?

> but stops working again if I try to power up the host.

Do you know when exactly ? before or after skiboot ? 

when it comes back, try to hammer it with a dd + md5sum. Here is what
I get :
```
root@palmetto:~# while [ 1 ]; do time dd if=/dev/mtd6 of=/tmp/pnor bs=64k; md5sum /tmp/pnor; done
512+0 records in
512+0 records out

real    0m4.402s
user    0m0.002s
sys     0m4.351s
cc5288e728c383c27f98d5a8dd13d4e1  /tmp/pnor
```

If it is stable, then it is a register setting issue. Most certainly 
the CE control. A safe setting for fast read is `000b0641`. You can set 
it manually to do some tests, register `0x1E630004`

skiboot should be using Dual IO for these chips
```
        /* Use 2READ */
        ct->ctl_read_val = (ct->ctl_read_val & 0x2000) |
                (0x03 << 28) | /* Dual IO */
                (0x0d << 24) | /* CE# width 3T */
                (0xbb << 16) | /* 2READ command */
                (0x00 <<  8) | /* HCLK/16 (optimize later) */
                (0x02 <<  6) | /* 2 bytes dummy cycle (8 clocks) */
                (0x01);        /* fast read */
```
and Linux should change it to Dual read (command 0x3B) :
```
root@palmetto:~# devmem 0x1E630004 32 
0x203B2641
```
To be noted that the 'CE# inactive pulse delays' are getting wiped out. 
That could be a problem for Linux. But not for skiboot which sets them.
Speed should be fine. But pflash also does some very fine tuning and sets 
the CE# inactive pulse delays, dual IO, etc. 

> LIBFLASH: Found chip Macronix MXxxL51235F size 64M erase granule: 4K
> LIBFLASH: Flash >16MB, enabling 4B mode...
> LIBFLASH: Enabling controller 4B mode...

could you dump the flash setting registers after booting Linux ? 

`md 0x1e630000` under u-boot.

> I can also reproduce this on palm8, which I thought was a production
[... 37 lines cut ...]
[    9.276629] 0x000000000000-0x000000060000 : "u-boot"
[    9.317502] 0x000000060000-0x000000080000 : "u-boot-env"
[    9.358416] 0x000000080000-0x0000004c0000 : "kernel"
[    9.399052] 0x0000004c0000-0x000001c00000 : "rofs"
[    9.439639] 0x000001c00000-0x000002000000 : "rwfs"
[    9.492145] aspeed-smc 1e630000.flash-controller: Using 50 MHz SPI frequency
[    9.499297] aspeed-smc 1e630000.flash-controller: control register: 00000300
[    9.499348] aspeed-smc 1e630000.flash-controller: default control register: 00000300
[    9.499867] aspeed-smc 1e630000.flash-controller: mx25l25635e (32768 Kbytes)
[    9.507000] aspeed-smc 1e630000.flash-controller: write control register: 00022302
[    9.507051] aspeed-smc 1e630000.flash-controller: read control register: 203b2341
[    9.514599] aspeed-smc 1e630000.flash-controller: AHB frequency: 192 MHz
[    9.540715] aspeed-smc 1e630000.flash-controller: Trying HCLK/5...
[    9.627548] aspeed-smc 1e630000.flash-controller:   * [00080000] 0 HCLK delay, 4ns DI delay : PASS
[    9.714473] aspeed-smc 1e630000.flash-controller:   * [00000000] 0 HCLK delay, 0ns DI delay : PASS
[    9.801833] aspeed-smc 1e630000.flash-controller:   * [00090000] 1 HCLK delay, 4ns DI delay : PASS
[    9.801895] aspeed-smc 1e630000.flash-controller:  * -> good is pass 1 [0x00000000]
[    9.801943] aspeed-smc 1e630000.flash-controller: Trying HCLK/4...
[    9.875213] aspeed-smc 1e630000.flash-controller:   * [00008000] 0 HCLK delay, 4ns DI delay : PASS
[    9.948482] aspeed-smc 1e630000.flash-controller:   * [00000000] 0 HCLK delay, 0ns DI delay : PASS
[   10.021970] aspeed-smc 1e630000.flash-controller:   * [00009000] 1 HCLK delay, 4ns DI delay : PASS
[   10.022022] aspeed-smc 1e630000.flash-controller:  * -> good is pass 1 [0x00000000]
[   10.022099] aspeed-smc 1e630000.flash-controller: Found good read timings at HCLK/4
[   10.097666] libphy: Fixed MDIO Bus: probed
```

shenki, 2018-08-14:
> what is the 0x1E630004 register setting before
>
> > it comes back,
>
> and after ?

It's working at the moment. Here is a machine that booted into the
host (so skiboot has run):

```
root@palm4-bmc:~# dmesg |grep smc
[    5.935486] aspeed-smc 1e620000.flash-controller: Using 50 MHz SPI frequency
[    5.942715] aspeed-smc 1e620000.flash-controller: mx25l25635e (32768 Kbytes)
[    5.949949] aspeed-smc 1e620000.flash-controller: CE0 window [
0x20000000 - 0x22000000 ] 32MB
[    5.958515] aspeed-smc 1e620000.flash-controller: CE1 window [
0x22000000 - 0x26000000 ] 64MB
[    5.967059] aspeed-smc 1e620000.flash-controller: read control
register: 203b0641
[    6.563759] aspeed-smc 1e630000.flash-controller: Using 50 MHz SPI frequency
[    6.570957] aspeed-smc 1e630000.flash-controller: mx66l51235l (65536 Kbytes)
[    6.578043] aspeed-smc 1e630000.flash-controller: read control
register: 203c2641
root@palm4-bmc:~# devmem 0x1E630004
0x3DBB2781
```

It tends to be stable once it's had a good boot. I will monitor it.

Another option is we end up moving to the "mbox"-managed flash, once
Andrew has finished routing it through ipmi. I want to benchmark it
first, as the existing mailbox register based method is much slower
than direct flash access for a normal boot.

In this scenario we would stop skiboot from touching the flash
controller all together.

shenki, 2018-08-14:
> root@palm4-bmc:~# devmem 0x1E630004
> 0x3DBB2781

0x3DBB2781 is a skiboot setting. 

When the host boots, skiboot overrides the OpenBMC Linux setting with   
settings like Dual IO and CE# inactive pulse delays.

Also, U-Boot (or Linux) should reset the SPI controller registers to 0x0. 
Else when the BMC (soft) resets, it inherits from the skiboot setting ... 
which could lead to some issues.

AlexanderAmelkin, 2018-08-14:
A very similar issue exists for YADRO VESNIN (P8) with the latest openbmc master, regardless of skiboot version.

AlexanderAmelkin, 2018-08-14:
An interesting fact is that we connected a logic analyzer directly to the PNOR flash chip and found out that while for the first boot attempt after AC on, there is an activity on the SPI bus (although boot fails), for subsequent power on attempts the SPI bus remains completely silent (CE# high, clock and DI/DO lines low).

nest1ing, 2018-08-14:
Some details about reproducing the problem on YADRO VESNIN (P8):

> We can't start the host on Vesnin with the latest OpenBMC after power cable unplugged.
> As workaround we use this steps:
> 1. Install old OpenBMC (kernel 4.10);
> 2. Install the latest OpenBMC.
> 3. Start the host.
> It work till we have power cable unplug, after this the host is unable to start at all.

Registers before trying to start the host (after power restored):
```
  0x1e789080: 0x0DBF0500
  0x1e789088: 0x30000C00
  0x1e78908c: 0xFC0003FF
  0x1E630004: 0x203C2041
  0x1E630014: 0x00000050
```
The Hostboot does not start, SOL is empty.

Registers after `watchdog timer was expired` messages in openbmc log:
```
  0x1e789080: 0x0DBF0500
  0x1e789088: 0x30000C00
  0x1e78908c: 0xFC0003FF
  0x1E630004: 0x00002400
  0x1E630014: 0x00000050
```

PNOR flash is unreadable now and at the next trying to start the host we have followed messages in log:
```
Aug 14 11:07:10 pflash[1321]: Error 100 opening ffs !
ug 14 11:07:10 pflash[1321]: Flash info:
Aug 14 11:07:10 pflash[1321]: -----------
Aug 14 11:07:10 pflash[1321]: Name          = /dev/mtd6
Aug 14 11:07:10 pflash[1321]: Total size    = 64MB         Flags E:ECC, P:PRESERVED, R:READONLY, B:BACKUP
Aug 14 11:07:10 pflash[1321]: Erase granule = 64KB           F:REPROVISION, V:VOLATILE, C:CLEARECC
Aug 14 11:07:10 pflash[1321]: FFS: Flash header not found. Code: 100
```

nest1ing, 2018-08-14:
> Yesterday I made reflashing cycle with old and new openbmc firmwares to test your patch from the gerrit about openbmc/openbmc#3357.
> The host is able to start. Kernel messages contain follow lines:
> ```
> [    6.992560] aspeed-smc 1e630000.flash-controller: Using 100 MHz SPI frequency
> [    7.000011] aspeed-smc 1e630000.flash-controller: mx66l51235l (65536 Kbytes)
> [    7.007104] aspeed-smc 1e630000.flash-controller: read control register: 2d3c2e41
> ```
> I rebuilt firmware with default values of spi-max-frequency and reflashed it without unplugging power cable:
> ```
> [    6.677457] aspeed-smc 1e630000.flash-controller: Using 50 MHz SPI frequency
> [    6.684660] aspeed-smc 1e630000.flash-controller: mx66l51235l (65536 Kbytes)
> [    6.691750] aspeed-smc 1e630000.flash-controller: read control register: 2d3c2641
> ```
> The host is still able to start. Then I unplugged and restored power:
> ```
> [    6.628246] aspeed-smc 1e630000.flash-controller: Using 50 MHz SPI frequency
> [    6.635469] aspeed-smc 1e630000.flash-controller: mx66l51235l (65536 Kbytes)
> [    6.642698] aspeed-smc 1e630000.flash-controller: read control register: 203c234
> [    6.854477] aspeed-smc 1e630000.flash-controller: No good frequency, using dumb slow
> ```

nest1ing, 2018-08-14:
And `pflash --direct -t -g` solve the problem until the power is unplugged.

AlexanderAmelkin, 2018-08-14:
Another interesting fact. We've added debug printout to `pflash` (into `ast_ahb_writel()` from arch_flash_arm.c), and found out that `pflash --direct -t -g` changes value of the following SPI-related registers: 0x1e630000, 0x1e630004, 0x1e630014. If however we set them manually to the same values that exist after `pflash`, then host is still unable to boot. Any thoughts?

legoater, 2018-08-14:
Try clearing all registers : 

`for i in 00 04 10 14; do devmem 0x1E6300$i 32 0x0; done`

and rerun ` pflash --direct -t -g`

See if you get some valid reads with : 

`while [ 1 ]; do time dd if=/dev/mtd6 of=/tmp/pnor bs=64k; md5sum /tmp/pnor; done`

or simply read the first bytes of  the flash device from the AHB window which is at  `0x30000000`

mine260309, 2018-08-15:
> Registers after watchdog timer was expired messages in openbmc log:
> 
>   0x1e789080: 0x0DBF0500
>   0x1e789088: 0x30000C00
>   0x1e78908c: 0xFC0003FF
>   0x1E630004: 0x00002400
>   0x1E630014: 0x00000050
> PNOR flash is unreadable now and at the next trying to start the host we have followed messages in log:

Does this also occur when `mbox` is removed from the build and replaced by `aspeed-lpc-ctrl`?
The reason I ask is because I want to know who changes the register `0x1E630004`, BMC or host?

legoater, 2018-08-15:
The CE control register  `0x1E630004` is changed by the Linux kernel SMC driver and by pflash on the BMC. On the host, skiboot changes it also through the LPC2AHB SIO device.

mine260309, 2018-08-15:
The issue Yadro met dose not involve skiboot, the hostboot fails to start.
From @shenki it is [SfcAST2400::hwInit()](https://github.com/open-power/hostboot/blob/fde2240fafb48866ef3324f57ae7f4417625ad18/src/usr/pnor/sfc_ast2400.C#L91) that does the setting.

legoater, 2018-08-15:
Yes. hostboot also does. the setting is a very basic and safe one. When the host does not boot, could you try clearing the read timing register ?

shenki, 2018-08-15:
Some notes:

1. I can reproduce this reliably by starting the process with a PDU (wall socket off) power cycle
2. Booting with `aspeed_smc.optimize_read=0` allows the BMC kernel to read the flash even after hostboot has run. Hostboot fails to start though, so we do not boot.
3. Booting with the out of tree spi-nor changes reverted shows the same behaviour as 2

@legoater which is the read timing register?

legoater, 2018-08-15:
`0x1e630014` is the read timing register.

I have noticed that the Linux kernel SMC driver on the BMC and skiboot do not have the same results for the read timing settings. Problably because the wires are not  the  same. But the skiboot settings are less permissive, so we should be fine on the BMC.

If you disable `aspeed_smc.optimize_read`, the BMC will not change the read timing settings anymore. It will only tune the CE Control Register. 

Are we seeing this problem only on P8 systems ?  What we good to know is if it is possible to boot the system after having restored the SPI controller registers to sane values.

shenki, 2018-08-15:
Yes, only on P8 as P9 systems do not touch the flash registers from the host (so hostboot nor skiboot modify settings).

I figured that doing `pflash --direct` was restoring things to a 'sane' state. From what you're saying libflash (aka skiboot and pflash) use a different idea of sane, so we should attempt a different set of values.

legoater, 2018-08-15:
real sane values are the default 0x0. 

pflash tries to get best setting but if the flash device has freaked out, it will get nothing out of it.

CPM80, 2018-08-15:
I've try to clean all SPI controller registers as you suggest. Next I've run `pflash --direct -t -g`. It return FAIL for all prescalers. Registers values are:
`0x1E630000 - 0x00000000`
`0x1E630004 - 0x3DBB2081`
`0x1E630014 - 0x00055550`
  Host was booted normally. Second run `pflash --direct -t -g` gived normal calibration results. Register values are:
`0x1E630000 - 0x00000000`
`0x1E630004 - 0x3DBB2781`
`0x1E630014 - 0x000009A0`
  And host was booted normally too. Does `pflash --direct -t -g` change anything except this registers?

shenki, 2018-08-15:
At boot:
```
0x1E630004 0x203C2041
0x1E630014 0x00005000
```
After an attempted (but failed) boot of the host:
```
0x1E630004 0x00002400
0x1E630014 0x00005000
```
I tried clearing `0x1E630014` and re-trying the boot and it did not work.

CPM80, 2018-08-15:
After host booted registers values are:
`0x1E630000 - 0x00000003`
`0x1E630004 - 0x00002400`
`0x1E630014 - 0x000009A0`

legoater, 2018-08-15:
`0x3DBB2081`, `0x3DBB2781` are set by skiboot (DUAL IO)
`0x203C2041` is set by the Linux kernel BMC (DUAL read)
`0x00002400` is set by hostboot

So after booting of the host, which means you have passed skiboot, a `0x00002400` value is curious. Unless you use a custom version of skiboot.

CPM80, 2018-08-15:
I've read register values very early. After host OS was loaded '0x1E630004' value is '0x3DBB2781'.

mine260309, 2018-08-15:
Some interesting info about the PNOR chip:
1. My Palmetto was using `n25q256a`, and it works fine with below log:
```
[    6.600612] aspeed-smc 1e630000.flash-controller: Using 50 MHz SPI frequency
[    6.607827] aspeed-smc 1e630000.flash-controller: n25q256a (32768 Kbytes)
[    6.614799] aspeed-smc 1e630000.flash-controller: read control register: 203b2341
```
2. Then I changed to another chip `n25q512ax3`, it works fine, too:
```
[    6.519326] aspeed-smc 1e630000.flash-controller: Using 50 MHz SPI frequency
[    6.526558] aspeed-smc 1e630000.flash-controller: n25q512ax3 (65536 Kbytes)
[    6.533556] aspeed-smc 1e630000.flash-controller: read control register: 203b2341
[    6.567184] aspeed-smc 1e630000.flash-controller: Calibration area too uniform, using low speed
```
3. Then I changed to `mx66l51235l`, which is the same type as Yadro and @shenki, the issue is reproduced:
```
[    6.519033] aspeed-smc 1e630000.flash-controller: Using 50 MHz SPI frequency
[    6.526246] aspeed-smc 1e630000.flash-controller: mx66l51235l (65536 Kbytes)
[    6.533330] aspeed-smc 1e630000.flash-controller: read control register: 203c2341
[    6.745008] aspeed-smc 1e630000.flash-controller: No good frequency, using dumb slow
```

So with three types of PNOR chip, only `mx66l51235l` had the issue.

nest1ing, 2018-08-15:
> Does this also occur when mbox is removed from the build and replaced by aspeed-lpc-ctrl?
> The reason I ask is because I want to know who changes the register 0x1E630004, BMC or host?
Yes, it was occurring with original mboxd, without mboxd, with aspeed-lpc-ctl.
I guess, this registers changed by host.

alvintpwang, 2018-08-15:
Compared the dev-4.10 and dev-4.17 and found mx66l51235l has SPI_NOR_4B_OPCODES flag set in dev-4.17. That's why the read op-code is different from n25q512ax3 to mx66l51235l. 

git diff  origin/dev-4.10 origin/dev-4.17  drivers/mtd/spi-nor/spi-nor.c | grep mx66l51235l
-	{ "mx66l51235l", INFO(0xc2201a, 0, 64 * 1024, 1024, SPI_NOR_DUAL_READ | SPI_NOR_QUAD_READ) },
+	{ "mx66l51235l", INFO(0xc2201a, 0, 64 * 1024, 1024, SPI_NOR_DUAL_READ | SPI_NOR_QUAD_READ | SPI_NOR_4B_OPCODES) },

mine260309, 2018-08-15:
Some more info:

1. With the `mx66l51235l` chip, BMC is able to read PNOR by default (after ac cycle) with below config:
```
devmem 0x1e630000 = 0x00000003
devmem 0x1e630004 = 0x203c2041
devmem 0x1e630014 = 0x00005000
```
2. When I manually set SPI04 to 0x00002400, BMC fails to read the PNOR. (Note that with other chips like `n25q256a`, this is still OK).
   This value is set by hostboot [SfcAST2400::hwInit()](https://github.com/open-power/hostboot/blob/fde2240fafb48866ef3324f57ae7f4417625ad18/src/usr/pnor/sfc_ast2400.C#L91), this explains why the host fails to boot.
3. After running `pflash --direct -t -g`, BMC is able to read PNOR again, and host is able to boot.

So in all, the setting `0x1e630004 = 0x00002400` and `0x1e630014 = 0x00005000` does not work with `mx66l51235l` chip.

Probably we should check what's the default setting in old OpenBMC?

AlexanderAmelkin, 2018-08-15:
Found out that aspeed-smc.c has timings register set to `0x94` for ast2400, which is wrong (there is no such register in SPI controller for AST2400, unlike FMC controller and SPI controller in AST2500).

Also, as @alvintpwang noticed above, the SPI_NOR_4B_OPCODES option is wrong for 0xc2201a chip, which is also wrongly identified as mx66l51235l. There is no such part number according to Macronix site, digikey.com. This part number can only be met in linux kernel and in u-boot. That's all. The actual Macronix chip part number for ID 0xc2201a is mx66l51235l.

If those two things are fixed, the host starts booting for us after AC cycle.

We've submitted a couple of patches to openbmc mailing list and to @shenki

shenki, 2018-08-16:
Thanks for tracking this down. I have merged the patches to dev-4.17 and pushed to github.

legoater, 2018-08-16:
I would be good to maintain a table of some kind with the different platforms and the flash device models they use by default, or which have been successfully tested.

shenki, 2018-08-17:
I started a table. I'm open to suggestions of moving it somewhere else if someone can think of a spot: https://github.com/openbmc/linux/wiki/SPI-Flash-Chips

nest1ing, 2018-08-17:
> The actual Macronix chip part number for ID 0xc2201a is mx66l51235l.

I guess @AlexanderAmelkin was mistyped, and correct part number is mx66l51235**f**

rfrandse, 2018-08-17:
<https://gerrit.openbmc-project.xyz/12012> kernel: ast2400 MTD, FSI ColdFire
Resolves:  #3358 p8: failing to read flash at boot

AlexanderAmelkin, 2018-08-28:
It looks like we were too fast on closing this. It appears that the bug is not on Linux side, but on SBE side instead. Please see http://lists.infradead.org/pipermail/linux-mtd/2018-August/083649.html

Our inspection of SBE code revealed that it expects the PNOR chip to be in EN4B mode, uses normal READ (0x03) command instead of READ4B (0x13), and doesn't enter permanent 4B mode (EN4B) itself. Thus, when BMC Linux stopped entering 4B mode, it broke flash reading for SBE.

I believe this must be fixed in SBE for P8. I guess I'll open a ticket for that in open-power organization. Once that bug is resolved in SBE, the patch to Linux will need to be reverted.

dcrowell77, 2018-08-28:
Repeating comment from the Hostboot issue:

The SBE code has to work on all systems, regardless of who/what is acting as the flash controller. This is because the code lives inside the processor module and cannot be updated when moved between systems. Any change here is likely to break other systems. Therefore the fix cannot come from the SBE side, the BMC needs to revert its behavior to support the existing hardware/firmware design.

The other note here is that the host code doesn't send any opcodes itself. In P8 we use logic inside the P8 processor to drive the PNOR operations. The hardware logic gives us a bit that says 'read', I don't see any other options in a quick skim.

AlexanderAmelkin, 2018-08-28:
@dcrowell77, thank you. We've already found that SBE code indeed doesn't use READ opcode directly and relies on BMC's SPI controller mapping to LPC bus.

@shenki, it looks like there is no other way than reverting locally in openbmc the change made by Linux commit [d342b6a973af](https://github.com/openbmc/linux/commit/d342b6a973af). Linux MTD maintainer is unlikely to accept this patch into upstream.

legoater, 2018-08-28:
@AlexanderAmelkin could you send a link to the SBE code please ?

AlexanderAmelkin, 2018-08-29:
@legoater, sure. It's public: [open-power/p8-sbe](https://github.com/open-power/p8-sbe)

mdmillerii, 2018-08-29:
@AlexanderAmelkin does it set the opcode or just use the existing read mapping?
Can you point to specific lines in the repository where it does a read?

AlexanderAmelkin, 2018-08-30:
@mdmillerii, I'm not sure about specific lines in repository, but it looks like it's somewhere within 
[proc_sbe_lco_loader.S](https://github.com/open-power/p8-sbe/blob/master/src/engd/href/eclipz/chips/p8/working/procedures/ipl/sbe/proc_sbe_lco_loader.S)

It doesn't look like it does anything at all with the SPI controller (including setting the opcode), it just uses the existing mapping. This corresponds to the position that @dcrowell77 expressed above, as well as to [P8_IPL_Flow_OpenPower.pdf](https://github.com/open-power/docs/blob/master/hostboot/P8_IPL_Flow_OpenPower.pdf). SBE relies on the existing mapping to be set up by the BMC.

AlexanderAmelkin, 2018-08-30:
@geissonator, maybe we should reopen this issue as it doesn't seem to have been fixed in a proper way?

legoater, 2018-08-30:
That is weird. The controller should then use the settings of the CE0 Control register to read PNOR. The SPI opcode 3C should be sent which is fine for 4BYTE addresses.

AlexanderAmelkin, 2018-08-30:
@legoater, hmm... Indeed weird. MX66F51235F does support opcode 3C. I thought that maybe hostboot is to blame because it sets mode 0x00002400 for SPI04 as @mine260309 mentioned earlier:

> When I manually set SPI04 to 0x00002400, BMC fails to read the PNOR. (Note that with other chips like n25q256a, this is still OK).
> This value is set by hostboot SfcAST2400::hwInit(), this explains why the host fails to boot.

However, after carefully checking hostboot code, we found that `SfcAST2400::hwInit()` is not called for Palmetto and VESNIN because they have `BMC_DOES_SFC_INIT` set. Still they fail to boot when BMC Linux leaves the PNOR flash in the default 3-byte (EX4B) mode. Hostboot also doesn't issue any opcodes for reading flash and uses the preconfigured mapping.

dcrowell77, 2018-08-30:
The first thing to note is that Hostboot is the *second* entity to access the PNOR during boot, the SBE is first.  The SBE uses a combination of built-in hardware access and a very thin layer of firmware to access the PNOR.  The hardware is immutable and the firmware must be compatible with every P8 system we've ever shipped, including those that use a FPGA as the SPI controller.  Note that the SBE only does reads, never erase/write.

As mentioned before, we don't send any opcodes at all for reads.  We do a MMIO to the LPC memory space and then hardware does the rest.  We do send explicit commands for erase and write.  We shouldn't be doing erases/writes on every boot, only if something changes in the system.

AlexanderAmelkin, 2018-08-30:
@dcrowell77, it's totally mysterious then why does commit [d342b6a973af](https://github.com/openbmc/linux/commit/d342b6a973af) to Linux kernel break booting for P8. I guess it's worth a try to replace legacy opcodes in hostboot for erase and write with 4-byte opcodes, and see if it fixes this issue without reverting the Linux commit.

In fact it would definitely be better if hostboot read SFDP and used appropriate opcodes for all systems. That would increase compatibility.

dcrowell77, 2018-08-30:
It is possible that some of our systems may use PNOR connections that don't support SDFP commands.  And it is possible that those were the systems that existed first...

AlexanderAmelkin, 2018-08-31:
@dcrowell77, well, then this (both support for SFDP and opcode for legacy systems) could be configured  via the system configuration file (op-build/openpower/configs/hostboot/*.config). A safe legacy-compatible default behavior can be chosen. However, this starts to look a lot like this discussion has to be moved back to [hostboot issue](https://github.com/open-power/hostboot/issues/146). Here we'll just need to revert https://gerrit.openbmc-project.xyz/12012 once this is fixed in hostboot.

dcrowell77, 2018-09-01:
We welcome patches...  At this point the Hostboot P8 codebase is in maintenance-only mode.

closed 2019-04-26 by stale[bot]

shenki, 2019-04-29:
@legoater did we arrive at a proper fix for this?

Will systems that use HIOMAP work around this bug?

mine260309, 2019-04-30:
Yes, I tested HIOMAP on Palmetto and it works fine.