openbmc/openbmc #3359: Response to CVE-2018-5390

joseph-reynolds, 2018-08-14:
This is my draft response to CVE-2018-5390 using draft guidelines in review: https://gerrit.openbmc-project.xyz/#/c/openbmc/docs/+/11665/3/security/obmc-security-response-team-guidelines.md.

This security vulnerability is not confidential, so our response is not confidential.
CVE-2018-5390 became public around 2018-08-06.  See https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2018-5390.

**OpenBMC releases affected**: (TODO: earliest through latest, TBD).  As of 2018-08-14 OpenBMC is on openbmc/import-layers/yocto-poky/meta/recipes-kernel/linux/linux-yocto_4.12.bb which is affected.  But note that various BSP recipes in the meta-openbmc-bsp set the LINUX_VERSION bitbake variable to different values.
TO DO: get earliest affected release.

**The fix** is to pick up a new kernel version which is typically done via Yocto.  TO DO: find out when Yocto will get this fix.  See https://wiki.yoctoproject.org/wiki/Security.

**CVSS score**: 7.5 (High).  My attempt to score it is here: https://www.first.org/cvss/calculator/3.0#CVSS:3.0/AV:N/AC:L/PR:N/UI:N/S:U/C:N/I:N/A:H/RC:C.

gtmills, 2018-08-14:
@joseph-reynolds, @shenki manages our kernel. It looks like we are at `4.17.7`.
https://github.com/openbmc/openbmc/blob/master/meta-openbmc-bsp/meta-aspeed/recipes-kernel/linux/linux-aspeed_git.bb

shenki, 2018-08-15:
Correct, we do not use the yocto kernel. At the moment we are on a 4.17.11 based tree, with additional patches.

This commit will move us up to 4.17.14, which includes the backported commits to net/ipv4/tcp_input.c.

joseph-reynolds, 2018-08-15:
Thanks for the information.  I see the following for repo openbmc/openbmc:
- meta-openbmc-bsp/meta-aspeed/recipes-kernel/linux/linux-aspeed_git.bb
   - LINUX_VERSION ?= "4.17.11" **(vulnerable)**
   - Moving to 4.17.14 **(fixed)** per https://gerrit.openbmc-project.xyz/+/11953/
-  meta-openbmc-bsp/meta-ibm/meta-fsp2/recipes-kernel/linux/linux-fsp2_4.13.bb 
   - LINUX_VERSION ?= "4.13.16" **(vulnerable)**
- meta-openbmc-bsp/meta-nuvoton/recipes-kernel/linux/linux-nuvoton_git.bb
   - LINUX_VERSION ?= "4.17.4" **(vulnerable)**

joseph-reynolds, 2018-08-15:
Not a kernel hacker, just trying to follow along at home...

I see a series of commits on Jul 23, 2018 that seem to address this CVE.  They include: https://github.com/torvalds/linux/commit/3d4bf93ac12003f9b8e1e2de37fe27983deebdcf#diff-d06665370d013ab0df986a88d8e3a128.
Is there a definitive source that says which commit addresses the CVE?  I didn't find it by searching the linux github repo or by googling the CVE with the fix.

rfrandse, 2018-08-15:
<https://gerrit.openbmc-project.xyz/11953> kernel: Move to v4.17.14 stable release
Resolves:  #3359 Response to CVE-2018-5390

joseph-reynolds, 2018-08-15:
Pinging apparent maintainers of Nuvaton @BenjaminFair @bradbishop and FSP2 @fr0st61te.

BenjaminFair, 2018-08-15:
I'll make a patch for the Nuvoton BSP bringing the kernel to 4.17.14