openbmc/openbmc #3360: net-snmp-server-snmpd build failure

nest1ing, 2018-08-15:
We are using the original `snmpd` from the `net-snmp` package in OpenBMC.
In our yocto-layer we added the `net-snmp-server-snmpd` to the `OBMC_IMAGE_EXTRA_INSTALL`

The commit 1baa4da706692781e9bef68fbbe72b576ba1c7e8 breaks build for us:
```
Error: 
 Problem: conflicting requests
  - nothing provides net-snmp-mibs needed by net-snmp-server-snmpd-5.7.3-r0.armv5e

ERROR: obmc-phosphor-image-1.0-r0 do_rootfs: Function failed: do_rootfs
```

nest1ing, 2018-08-15:
I suggest to split the `net-snmp-libs` into smaller packages and use them if required as a dependency.
I push my solution to https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/11964

rfrandse, 2018-08-15:
<https://gerrit.openbmc-project.xyz/11964> split net-snmp-libs into smaller packages
Resolves  #3360 net-snmp-server-snmpd build failure

closed 2018-09-21 by geissonator