openbmc/openbmc #3362: How bitbake chooses which recipes need to be compiled

fengyd2018, 2018-08-19:
1. A new layer is added with following commands
. setup evb-ast2500
bitbake-layers create-layer ../my_layer
bitbake-layers add-layer ../my_layer
2. The recipes in the new layer aren't compile when running bitbake obmc-phosphor-image

How bitbake chooses which recipes need to be compiled?

anoo1, 2018-08-20:
@bradbishop can help with this question. @fengyd2018 feel free to post your questions on IRC as well (freenode channel #openbmc).

gtmills, 2018-08-20:
@fengyd2018 This is a good question for the mailing list or IRC, as it's not a bug or feature request. You can sign yourself up for the mailing list at https://lists.ozlabs.org/listinfo/openbmc. Thanks!

closed 2019-01-25 by anoo1