openbmc/openbmc #3364: Add -flto (Link Time Optimization) to C++ compiles to reduce size

geissonator, 2018-08-20:
This may be a quick and easy way to shave down the size of our C/C++ binaries and libraries.  This came up as an idea in todays community call by @edtanous who said it helped quite a bit in bmcweb.

An easy fix would be to just add it to one of the CFLAGS/CXXFLAGS already used by most repos (like SDBUSPLUS_CFLAGS) and then let repos override that if there's a need.

http://gcc.gnu.org/wiki/LinkTimeOptimization

geissonator, 2018-08-23:
Tried blindly adding it to all C++ compiles via import-layers/yocto-poky/meta/conf/bitbake.conf, gperf didn't like it:

```
| /tmp/cc6HxoxX.ltrans0.ltrans.o: In function `main':
| <artificial>:(.text.startup+0xae8): undefined reference to `get_delim(char**, unsigned long*, int, _IO_FILE*)'
| /tmp/cc6HxoxX.ltrans7.ltrans.o: In function `Hash_Table::insert(KeywordExt*) [clone .constprop.6]':
| <artificial>:(.text+0x233): undefined reference to `hashpjw(unsigned char const*, unsigned int)'
| collect2: error: ld returned 1 exit status
| Makefile:74: recipe for target 'gperf' failed
| make[1]: *** [gperf] Error 1
| make[1]: Leaving directory '/home/andrewg/Code/openbmc/build/tmp/work/x86_64-linux/gperf-native/3.1-r0/build/src'
| ERROR: oe_runmake failed
| Makefile:33: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/andrewg/Code/openbmc/build/tmp/work/x86_64-linux/gperf-native/3.1-r0/temp/log.do_compile.30329)
ERROR: Task (virtual:native:/home/andrewg/Code/openbmc/meta/recipes-extended/gperf/gperf_3.1.bb:do_compile) failed with exit code '1'
```

geissonator, 2018-08-27:
After some discussion, was decided that each repo should add this (not use a global across all repos).  I put up some commits with the top size impacts with https://gerrit.openbmc-project.xyz/#/q/topic:flto+(status:open+OR+status:merged).

gtmills, 2018-10-12:
@geissonator https://gerrit.openbmc-project.xyz/#/q/topic:flto+(status:open+OR+status:merged) are all merged. Can this be closed?

geissonator, 2019-02-25:
Yeah, we've done as much as we're going to do here. It's up to repo maintainers to add when appropriate.

closed 2019-02-25 by geissonator