openbmc/openbmc #3365: Reduce openbmc flash size footprint

geissonator, 2018-08-20:
A general epic to track different issues for reducing the overall footprint of openbmc in regards to flash space usage. Although openbmc is not required to fit in any specific size storage, a lot of use cases are in small embedded systems so anything we can do to reduce overall size without reducing function or options for bigger images is goodness.

geissonator, 2018-08-20:
Allowing system owners to remove python is also another option.  This for the most part would fall under https://github.com/openbmc/openbmc/issues/262 where the python applications in skeleton are being refactored into C++ applications.

geissonator, 2018-08-22:
[dirsize-100k-rootfs-sort.log](https://github.com/openbmc/openbmc/files/2311997/dirsize-100k-rootfs-sort.log)

Here's a quick use of dirsize.py within a witherspoon build to show everything that goes into the rootfs.  It grabbed everything greater then 100K in size.

geissonator, 2018-08-22:
Some big hitters to think about:

/usr/lib/python2.7
/usr/lib/libpython2.7.so.1.0
/usr/share/www/redfish/v1/schema
/usr/lib/ipmid-providers
/usr/lib/ipmid-providers/libipmi20.so.0.0.0

edtanous, 2018-09-07:
Patchset here should save us ~600KB in phosphor-webui.

https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-webui/+/12565/

edtanous, 2018-11-01:
FYI, just built a tool that might help us find good candidates for things to pull out of the rootfs and did a quick analysis of my development image.
https://gerrit.openbmc-project.xyz/#/c/openbmc/openbmc-tools/+/15142/

geissonator, 2019-02-25:
We can open specific stories in the future as they come up for this. We're now running the above tool in all openbmc builds (https://openpower.xyz/job/openbmc-build/distro=ubuntu,label=builder,target=witherspoon/lastSuccessfulBuild/artifact/deploy/images/results.txt). Overall we're moved the needle in the right direction, and with the ability to optionally include python coming, we'll be even better.

closed 2019-02-25 by geissonator