openbmc/openbmc #3366: Support for IPMI over LAN

ajithadapa, 2018-08-21:
Does OPENBMC support IPMI interface over LAN in Wedge-100 ?

I do have wedge-100 switch running latest version of openbmc. I want to confirm if there is any IPMI interface or I need to stick with existing REST interface ?

ajithadapa, 2018-08-21:
As per my understanding this is not supported as per the comment in below issue 

https://github.com/openbmc/ipmitool/issues/3

geissonator, 2018-08-21:
https://github.com/openbmc/phosphor-host-ipmid/blob/master/docs/ipmitool-commands-cheatsheet.md

It supports the "lanplus" option if that's what you're asking?

ajithadapa, 2018-08-21:
Thanks for the reply @geissonator. But when I try to run ipmitool on BMC terminal it throws error saying invalid command. Do I need to install ipmitool ?

root@bmc:~# ipmitool
-bash: ipmitool: command not found
root@bmc:~# cat /etc/version 
201711091410
root@bmc:~# 
root@bmc:~# uname -a 
Linux bmc 2.6.28.9 #1 Thu Nov 9 14:45:46 PST 2017 armv5tejl GNU/Linux
root@bmc:~#

geissonator, 2018-08-21:
Yes, you need to install it.  You should probably try and use the one that comes with the openbmc distro - https://github.com/openbmc/ipmitool

gx0128, 2018-08-30:
@geissonator Could you please introduce how to install the ipmitool from https://github.com/openbmc/ipmitool? And how to use it in detail? Could you please give an example?Thanks

jinneal, 2018-09-04:
I install AST2500 and start it by QEMU, and I can execute the ipmitool by lanplus interface, but I can not execute by openipmi or KCS mode
root@openbmc:~# ipmitool -I lanplus -H 127.0.0.1 -P 0penBmc lan print
Set in Progress         : Set Complete
Auth Type Support       : MD5 
Auth Type Enable        : Callback : MD5 
                        : User     : MD5 
                        : Operator : MD5 
                        : Admin    : MD5 
                        : OEM      : MD5 
IP Address Source       : DHCP Address
IP Address              : 10.0.2.15
Subnet Mask             : 255.255.255.0
MAC Address             : 26:fb:64:4e:48:b1
Default Gateway IP      : 10.0.2.2
802.1q VLAN ID          : Disabled
RMCP+ Cipher Suites     : 3,17
Cipher Suite Priv Max   : Not Available
Bad Password Threshold  : Not Available

root@openbmc:~# ipmitool                                             
Could not open device at /dev/ipmi0 or /dev/ipmi/0 or /dev/ipmidev/0: No such file or directory

If I need to let the kernel support the drivers of IPMI?

edtanous, 2018-09-05:
Seeing as how this project doesn't have a Wedge100 configuration, I'm going to guess that you need to direct your question to the project here: https://github.com/facebook/openbmc