openbmc/openbmc #3367: How to config sensor for a float value?

lockless, 2018-08-22:
Hello,

I config the sensor in ...ipmi/phosphor-ipmi-sensor-inventory%/config.yaml like this:
```
0x40:
  sensorType: 0x04
  path: /xyz/openbmc_project/sensors/fan_tach/fan1
  sensorReadingType: 0x01
  scale: 0
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.RPMS
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: int64_t
0x41:
  sensorType: 0x04
  path: /xyz/openbmc_project/sensors/fan_tach/fan2
  sensorReadingType: 0x01
  scale: 0
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.RPMS
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: int64_t
0x42:
  sensorType: 0x04
  path: /xyz/openbmc_project/sensors/fan_tach/fan3
  sensorReadingType: 0x01
  scale: 0
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.RPMS
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: int64_t
0x43:
  sensorType: 0x04
  path: /xyz/openbmc_project/sensors/fan_tach/fan4
  sensorReadingType: 0x01
  scale: 0
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.RPMS
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: int64_t
0x44:
  sensorType: 0x01
  path: /xyz/openbmc_project/sensors/temperature/lm75out
  sensorReadingType: 0x01
  scale: -3
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.DegreesC
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: int64_t
0x45:
  sensorType: 0x01
  path: /xyz/openbmc_project/sensors/temperature/lm75hot
  sensorReadingType: 0x01
  scale: -3
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.DegreesC
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: int64_t
0x46:
  sensorType: 0x01
  path: /xyz/openbmc_project/sensors/temperature/lm75in
  sensorReadingType: 0x01
  scale: -3
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.DegreesC
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: Double
0x47:
  sensorType: 0x01
  path: /xyz/openbmc_project/sensors/temperature/psu1_temp
  sensorReadingType: 0x01
  scale: -3
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.DegreesC
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: Double
0x48:
  sensorType: 0x02
  path: /xyz/openbmc_project/sensors/voltage/psu1_vol_out
  sensorReadingType: 0x01
  scale: -3
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.Volts
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: Double
0x49:
  sensorType: 0x02
  path: /xyz/openbmc_project/sensors/voltage/psu1_vol_in
  sensorReadingType: 0x01
  scale: -3
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.Volts
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: int64_t
0x4A:
  sensorType: 0x08
  path: /xyz/openbmc_project/sensors/power/psu1_power_out
  sensorReadingType: 0x01
  scale: -3
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.Watts
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: Double
0x4B:
  sensorType: 0x08
  path: /xyz/openbmc_project/sensors/power/psu1_power_in
  sensorReadingType: 0x01
  scale: -3
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.Watts
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: int64_t
0x4C:
  sensorType: 0x03
  path: /xyz/openbmc_project/sensors/current/psu1_curr_out
  sensorReadingType: 0x01
  scale: -3  
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.Amperes
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: int64_t
0x4D:
  sensorType: 0x03
  path: /xyz/openbmc_project/sensors/current/psu1_curr_in
  sensorReadingType: 0x01
  scale: -3
  serviceInterface: org.freedesktop.DBus.Properties
  readingType: readingData
  unit: xyz.openbmc_project.Sensor.Value.Unit.Amperes
  mutability: Mutability::Write|Mutability::Read
  interfaces:
    xyz.openbmc_project.Sensor.Value:
      Value:
        Offsets:
          0x0:
            type: int64_t
```
and, when the device power, i can get sdr info like this

```
root@Ubuntu:/# ipmitool -I lanplus -H 192.168.23.22 -P 0penBmc sdr elist
fan1             | 40h | ok  |  0.0 | 228 revolutions
fan2             | 41h | ok  |  0.0 | 228 revolutions
fan3             | 42h | ok  |  0.0 | 228 revolutions
fan4             | 43h | ok  |  0.0 | 228 revolutions
lm75out          | 44h | ok  |  0.0 | 31 degrees C
lm75hot          | 45h | ok  |  0.0 | 49 degrees C
lm75in           | 46h | ok  |  0.0 | 47 degrees C
psu1_temp        | 47h | ok  |  0.0 | 34 degrees C
psu1_vol_out     | 48h | ok  |  0.0 | 12 Volts
psu1_vol_in      | 49h | ok  |  0.0 | 236 Volts
psu1_power_out   | 4Ah | ok  |  0.0 | 124 Watts
psu1_power_in    | 4Bh | ok  |  0.0 | 139 Watts
psu1_curr_out    | 4Ch | ok  |  0.0 | 10 Amps
psu1_curr_in     | 4Dh | ok  |  0.0 | 0 Amps
```
the question is , some value should be float/double
like this
```
          psu1_current_in : 0.589 A
          psu1_vol_in : 233.0 V
         psu1_curr_out : 10.234 A
         psu1_vol_out : 12.167 V
```
so, how to config to get this ?
thanks for advise!!!

gtmills, 2018-08-29:
@lockless this is a good question for the mailing list, as it's not a bug or feature request. You can sign yourself up at https://lists.ozlabs.org/listinfo/openbmc

Thanks!

gtmills, 2018-08-29:
@tomjoseph83 Know this one?

gtmills, 2018-10-12:
Closing. Not a bug or feature request. Please use the mailing list or IRC for questions like this. thanks!!

closed 2018-10-12 by gtmills