openbmc/openbmc #3370: Target machine can't be set to evb-ast2500

mashuo1981, 2018-08-24:
When I input “source setup evb-ast2500”
“No such machine!” was printout.

#source setup
Target machine must be specified. Use one of:

barreleye
centriq2400-rep
fsp2
lanyang
msn
neptune
palmetto
q71l
qemuarm
romulus
s2600wf
witherspoon
z
zaius

mashuo1981, 2018-08-24:
I

gtmills, 2018-08-24:
https://github.com/openbmc/openbmc/blob/master/setup is the setup script. 
Looks like it needs an update to find https://github.com/openbmc/openbmc/tree/master/meta-evb/meta-evb-aspeed/meta-evb-ast2500

mashuo1981, 2018-08-26:
Thanks for your reply, I using below command to compiler the IMG for ast500:

#export TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf
#. openbmc-env
#bitbake obmc-phosphor-image

closed 2018-08-26 by mashuo1981