openbmc/openbmc #3371: Can't access redfish by https.

mashuo1981, 2018-08-26:
I have added bmcweb in openbmc image, and I can access redfish by http, "curl -k -u root:0penBmc -H content-type:application/json -X GT https://127.0.0.1:8080/redfish/v1/", the port 8080 was forword to 883.
When I used 2443 port(forward to 443), the error was printed as below:
#curl -k -u root:0penBmc -H content-type:application/json -X GT https://127.0.0.1:2443/redfish/v1/
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /redfish/v1"
  },
  "message": "404 Not Found",
  "status": "error"
}

Addition: I can access restful by https, such as "curl -b cjar -k https://${bmc}/xyz/openbmc_project/".

gtmills, 2018-08-29:
@mashuo1981 What system is this? 
Is bmcweb included in the image?

mashuo1981, 2018-08-30:
@gtmills The system is ast-2500, and bmcweb is included in the image.

gtmills, 2018-09-04:
@geissonator or @edtanous Could you help with this one? Thanks!

edtanous, 2018-09-05:
Having some trouble understanding what's going on here.  Given the way bmcweb is architected, it will never return the response you're seeing, but phosphor-rest will.  What makes you think that port 8080 points to bmcweb? Have you modified the socket file to move it to 8080?  bmcweb by default responds to port 443, with HTTPS connections, although with changes to the build flags, can be configured to respond to HTTP connections on any port you choose.

I would start by making sure it works the way it's intended, not remoted over an SSH tunnel;  Once that works, then we can attempt to debug the SSH tunnel.

Can you push a patchset with the edits you've made so we can look over them?

mashuo1981, 2018-09-05:
@edtanous , I used Qemu to emulation AST2500 chip, and executed below command to run flahs image with qemu "/root/qemu-aspeed/arm-softmmu/qemu-system-arm -M ast2500-evb -m 512 \
-drive file=/root/openbmc/build/tmp/deploy/images/evb-ast2500/obmc-phosphor-image-evb-ast2500-20180826103424.static.mtd,format=raw,if=mtd \
-nodefaults -nographic -serial stdio -net nic \
-net user,hostfwd=::2222-:22,hostfwd=::2443-:443,hostfwd=::8080-:883,hostname=qemu"
The 8080 port was forwarded to 883, and 2443 port was forwarded to 443.

I didn't modify the source code, I just added IMAGE_INSTALL_append = "bmcweb" to build/conf/local.conf

edtanous, 2018-09-05:
I suspect what's happening is that you didn't remove phosphor-rest, which also hosts itself on port 443 by default.  Because we have SO_REUSEADDR set on both sides, each daemon could be fielding your request.

Try removing phosphor-rest from your image, or bbappending a socket file that moves it to a port other than 443, and try again.

edtanous, 2018-09-11:
Is your issue resolved?

gtmills, 2018-10-12:
Closing. If still an issue please reopen. Thanks!

closed 2018-10-12 by gtmills