openbmc/openbmc #3372: Remove .py or .pyc files from flash image

geissonator, 2018-08-27:
Ideally, remove pyc so that we can easily patch and view python files.  Only gotcha would be if this has major performance impacts.  If there are major perf impacts then we'll need to remove the .py files.

geissonator, 2018-08-27:
Removing the .pyc files added 30 seconds to the reboot time.  It also caused a noticeable lag when running the first REST command (starts up gevent).  Removing the .py files was not well received by the team.  May need to just continue working on removing the existing python apps.

geissonator, 2018-08-30:
I've adapted this to only remove the .py files for the python library itself. There's no need for these in our filesystem and removing them saves us 800KB of rofs space.  The other python apps written in the phosphor layer will still have the .py files.

rfrandse, 2018-08-31:
<https://gerrit.openbmc-project.xyz/12232> Reduce python.bb .py files
Resolves:  #3372 Remove .py files from python recipe in flash image