openbmc/openbmc #3373: GUI: "Select All" checkbox checks should select only filtered events

BeccaBroek, 2018-08-28:
Currently when a user selects all on the Event Log page, all events are selected, regardless of the filter. 

For example, if there are 3 low priority events:
[image]

And when select all input is selected while the low priority filter is applied,
[image]

51 events are selected. (All of the events) and subsequently are all deleted, exported, etc.

BeccaBroek, 2018-08-30:
Also seeing that when a filter is applied, and all boxes are checked, the 'select all' box is not checked as it should be.

[image]

rfrandse, 2018-08-30:
<https://gerrit.openbmc-project.xyz/12228> Event log select all checkbox
Resolves  #3373 GUI: "Select All" checkbox should select only filtered events

closed 2018-08-31 by geissonator