openbmc/openbmc #3375: Failed building obmc-phosphor-image

awxhw, 2018-09-07:
I hit memory exhaustion,
    "/work/armv5e-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+a38b0b2063-r0/git/src/webserver_main.cpp
| virtual memory exhausted: Cannot allocate memory
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/vagrant/source/tmp/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+a38b0b2063-r0/temp/log.do_compile.28770)
ERROR: Task (/home/vagrant/source/tmp/openbmc/meta-phosphor/common/recipes-phosphor/interfaces/bmcweb.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3205 tasks of which 3203 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/vagrant/source/tmp/openbmc/meta-phosphor/common/recipes-phosphor/interfaces/bmcweb.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code
"

geissonator, 2018-09-30:
I'm hitting this as well with a fresh build on Romulus with 2d622cb4f03ff0183cea051480d539a56acacb44

Any ideas @anoo1 or @bradbishop ?
```
ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_generate_static(d)
     0003:                                                                                                                                                                                                      
File: '/home/andrewg/Code/openbmc/meta-phosphor/classes/image_types_phosphor.bbclass', lineno: 212, function: do_generate_static                                                                                
     0208:                                                                                                                                                                                                      
     0209:    _append_image(os.path.join(d.getVar('DEPLOY_DIR_IMAGE', True),                                                                                                                                    
     0210:                               'u-boot.%s' % d.getVar('UBOOT_SUFFIX',True)),                                                                                                                          
     0211:                  int(d.getVar('FLASH_UBOOT_OFFSET', True)),                                                                                                                                          
 *** 0212:                  int(d.getVar('FLASH_KERNEL_OFFSET', True)))                                                                                                                                         
     0213:                                                                                                                                                                                                      
     0214:    _append_image(os.path.join(d.getVar('DEPLOY_DIR_IMAGE', True),                                                                                                                                    
     0215:                               d.getVar('FLASH_KERNEL_IMAGE', True)),                                                                                                                                 
     0216:                  int(d.getVar('FLASH_KERNEL_OFFSET', True)),
File: '/home/andrewg/Code/openbmc/meta-phosphor/classes/image_types_phosphor.bbclass', lineno: 200, function: _append_image
     0196:    nor_image = os.path.join(d.getVar('IMGDEPLOYDIR', True),
     0197:                             '%s.static.mtd' % d.getVar('IMAGE_NAME', True))
     0198:
     0199:    def _append_image(imgpath, start_kb, finish_kb):
 *** 0200:        imgsize = os.path.getsize(imgpath)
     0201:        if imgsize > (finish_kb - start_kb) * 1024:
     0202:            bb.fatal("Image '%s' is too large!" % imgpath)
     0203:
     0204:        subprocess.check_call(['dd', 'bs=1k', 'conv=notrunc',
File: '/usr/lib/python3.5/genericpath.py', lineno: 50, function: getsize
     0046:
     0047:
     0048:def getsize(filename):
     0049:    """Return the size of a file, reported by os.stat()."""
 *** 0050:    return os.stat(filename).st_size
     0051:
     0052:
     0053:def getmtime(filename):
     0054:    """Return the last modification time of a file, reported by os.stat()."""
Exception: FileNotFoundError: [Errno 2] No such file or directory: '/home/andrewg/Code/openbmc/build/tmp/deploy/images/romulus/u-boot.bin'

ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Function failed: do_generate_static
ERROR: Logfile of failure stored in: /home/andrewg/Code/openbmc/build/tmp/work/romulus-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.9269
ERROR: Task (/home/andrewg/Code/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4218 tasks of which 4212 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/andrewg/Code/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

````
andrewg@andrewg-VirtualBox:~/Code/openbmc/build$ ls /home/andrewg/Code/openbmc/build/tmp/deploy/images/romulus/
aspeed-bmc-opp-romulus.dtb                                                                       obmc-phosphor-image-romulus-20180930185507.testdata.json
fitImage                                                                                         obmc-phosphor-image-romulus.testdata.json
fitImage--4.18.8+git0+fd8f1293c6-r0-romulus-20180930161848.bin                                   obmc-phosphor-initramfs-romulus-20180930185507.rootfs.cpio.lzma
fitImage-its-4.18.8+gitAUTOINC+fd8f1293c6-r0-romulus-20180930161848.its                          obmc-phosphor-initramfs-romulus-20180930185507.rootfs.manifest
fitImage-its-obmc-phosphor-initramfs-romulus-4.18.8+gitAUTOINC+fd8f1293c6-r0-20180930161848.its  obmc-phosphor-initramfs-romulus-20180930185507.testdata.json
[... cut ...]
fitImage-obmc-phosphor-initramfs-romulus-4.18.8+gitAUTOINC+fd8f1293c6-r0-20180930161848.bin      zImage--4.18.8+git0+fd8f1293c6-r0-aspeed-bmc-opp-romulus-20180930161848.dtb
fitImage-obmc-phosphor-initramfs-romulus.bin                                                     zImage--4.18.8+git0+fd8f1293c6-r0-romulus-20180930161848.bin
fitImage-romulus.bin                                                                             zImage-aspeed-bmc-opp-romulus.dtb
obmc-phosphor-debug-tarball-romulus-20180930185507.testdata.json                                 zImage-romulus.bin
obmc-phosphor-debug-tarball-romulus.testdata.json
```

geissonator, 2018-09-30:
Workaround is to just run this first:
```
bitbake u-boot-aspeed
```
Then run the full bitbake obmc-phosphor-image

Must be a missing dependency on u-boot-aspeed.

osresearch, 2018-10-11:
Had the same problem with a clean checkout of c3966252f9945108d27845e12b7732a825a9ca4e; the same workaround worked for me:

```
bitbake u-boot-aspeed && bitbake obmc-phosphor-image
```

anoo1, 2018-10-11:
Some similar issues:
https://github.com/openbmc/openbmc/issues/3073
https://github.com/openbmc/openbmc/issues/3208

ErikvdBogaert, 2018-11-07:
Hi,

I am new here but I fixed this problem by replacing the line
```
DEPENDS += "dtc-native"
```
with
```
DEPENDS += "dtc-native u-boot-mkimage-native"
```

in the file: `openbmc\meta-aspeed\recipes-bsp\u-boot\u-boot-aspeed_2016.07.bb`

Can somebody tell me if this is the proper way to do it?

anoo1, 2019-01-25:
Reopen if issue still exist in the latest master.

anoo1, 2019-07-08:
Seems this is still an issue: https://gist.github.com/sotaoverride/b39579154d2bd376e5ca2510444c64dc

```
ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: [Errno 2] No such file or directory: '/home/aatir/openbmc/build/tmp/deploy/images/palmetto/u-boot.bin'
ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Function failed: do_generate_static
ERROR: Logfile of failure stored in: /home/aatir/openbmc/build/tmp/work/palmetto-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.13640
ERROR: Task (/home/aatir/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4109 tasks of which 10 didn't need to be rerun and 1 failed.
```

morchaos, 2019-08-17:
Same issue happened on me when building OpenBMC for Raspberry PI. the workaround did not work for me since there is no uboot for Raspberry pi. should I just try to increase the size or is there any better solution for this?

morchaos, 2019-08-17:
> Same issue happened on me when building OpenBMC for Raspberry PI. the workaround did not work for me since there is no uboot for Raspberry pi. should I just try to increase the size or is there any better solution for this?

by the way, I am using Ubuntu 18.04 for the build as well

closed 2020-03-17 by stale[bot]

Samar2610, 2023-03-09:
The is issue occuring in ubuntu 18 and was resolved in 
   Ubuntu 22.04 , 
   B.B version 2.2.0 ,
   Kernal version 5.19.0-35-generic 
and also with these prerequisite we must install some important libraries like libfdt and libslirp  and as it's not occurring on the latest image so we are closing this from our end.

mahendra-space, 2025-04-23:
ERROR: /home/vivek/gitlab/openBMC/openbmc/meta-phosphor/recipes-phosphor/watchdog/phosphor-watchdog_git.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 16, function: <module>
     0012:__anon_177__home_vivek_gitlab_openBMC_openbmc_meta_classes_siteinfo_bbclass(d)
     0013:__anon_155__home_vivek_gitlab_openBMC_openbmc_meta_phosphor_classes_obmc_phosphor_dbus_service_bbclass(d)
     0014:__anon_23__home_vivek_gitlab_openBMC_openbmc_meta_classes_systemd_bbclass(d)
     0015:__anon_195__home_vivek_gitlab_openBMC_openbmc_meta_classes_useradd_bbclass(d)
 *** 0016:__anon_239__home_vivek_gitlab_openBMC_openbmc_meta_phosphor_classes_obmc_phosphor_systemd_bbclass(d)
File: '/home/vivek/gitlab/openBMC/openbmc/meta-phosphor/classes/obmc-phosphor-systemd.bbclass', lineno: 230, function: __anon_239__home_vivek_gitlab_openBMC_openbmc_meta_phosphor_classes_obmc_phosphor_systemd_bbclass
     0226:        tmpl = [SystemdUnit(x) for x in tmpl]
     0227:        svc = [x for x in svc if not x.is_instance]
     0228:
     0229:        for unit in tmpl + svc:
 *** 0230:            unit_exist = check_sd_unit(d, unit)
     0231:            add_sd_unit(d, unit, pkg, unit_exist)
     0232:            add_sd_user(d, unit.name, pkg)
     0233:        for name in listvar_to_list(d, 'SYSTEMD_ENVIRONMENT_FILE_%s' % pkg):
     0234:            add_env_file(d, name, pkg)
File: '/home/vivek/gitlab/openBMC/openbmc/meta-phosphor/classes/obmc-phosphor-systemd.bbclass', lineno: 117, function: check_sd_unit
     0113:            # Unit does not exist in tree. Allow it to install from repo.
     0114:            # Return False here to indicate it does not exist.
     0115:            return False
     0116:
 *** 0117:        parser = systemd_parse_unit(d, path)
     0118:        inhibit = listvar_to_list(d, 'INHIBIT_SYSTEMD_RESTART_POLICY_WARNING')
     0119:        if unit.is_service and \
     0120:                not unit.is_template and \
     0121:                unit.name not in inhibit and \
File: '/home/vivek/gitlab/openBMC/openbmc/meta-phosphor/classes/obmc-phosphor-systemd.bbclass', lineno: 102, function: systemd_parse_unit
     0098:
     0099:
     0100:def systemd_parse_unit(d, path):
     0101:    import configparser
 *** 0102:    parser = configparser.SafeConfigParser(strict=False)
     0103:    parser.optionxform = str
     0104:    parser.read('%s' % path)
     0105:    return parser
     0106:
Exception: AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?

ERROR: /home/vivek/gitlab/openBMC/openbmc/meta-phosphor/recipes-phosphor/video/fbterm_git.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 15, function: <module>
     0011:__anon_20__home_vivek_gitlab_openBMC_openbmc_meta_classes_blacklist_bbclass(d)
     0012:__anon_177__home_vivek_gitlab_openBMC_openbmc_meta_classes_siteinfo_bbclass(d)
     0013:__anon_23__home_vivek_gitlab_openBMC_openbmc_meta_classes_systemd_bbclass(d)
     0014:__anon_195__home_vivek_gitlab_openBMC_openbmc_meta_classes_useradd_bbclass(d)
 *** 0015:__anon_239__home_vivek_gitlab_openBMC_openbmc_meta_phosphor_classes_obmc_phosphor_systemd_bbclass(d)
File: '/home/vivek/gitlab/openBMC/openbmc/meta-phosphor/classes/obmc-phosphor-systemd.bbclass', lineno: 230, function: __anon_239__home_vivek_gitlab_openBMC_openbmc_meta_phosphor_classes_obmc_phosphor_systemd_bbclass
     0226:        tmpl = [SystemdUnit(x) for x in tmpl]
     0227:        svc = [x for x in svc if not x.is_instance]
     0228:
     0229:        for unit in tmpl + svc:
 *** 0230:            unit_exist = check_sd_unit(d, unit)
     0231:            add_sd_unit(d, unit, pkg, unit_exist)
     0232:            add_sd_user(d, unit.name, pkg)
     0233:        for name in listvar_to_list(d, 'SYSTEMD_ENVIRONMENT_FILE_%s' % pkg):
     0234:            add_env_file(d, name, pkg)
File: '/home/vivek/gitlab/openBMC/openbmc/meta-phosphor/classes/obmc-phosphor-systemd.bbclass', lineno: 117, function: check_sd_unit
     0113:            # Unit does not exist in tree. Allow it to install from repo.
     0114:            # Return False here to indicate it does not exist.
     0115:            return False
     0116:
 *** 0117:        parser = systemd_parse_unit(d, path)
     0118:        inhibit = listvar_to_list(d, 'INHIBIT_SYSTEMD_RESTART_POLICY_WARNING')
     0119:        if unit.is_service and \
     0120:                not unit.is_template and \
     0121:                unit.name not in inhibit and \
[... 21 lines cut ...]
     0016:__anon_62__home_vivek_gitlab_openBMC_openbmc_meta_classes_allarch_bbclass(d)
File: '/home/vivek/gitlab/openBMC/openbmc/meta-phosphor/classes/obmc-phosphor-systemd.bbclass', lineno: 230, function: __anon_239__home_vivek_gitlab_openBMC_openbmc_meta_phosphor_classes_obmc_phosphor_systemd_bbclass
     0226:        tmpl = [SystemdUnit(x) for x in tmpl]
     0227:        svc = [x for x in svc if not x.is_instance]
     0228:
     0229:        for unit in tmpl + svc:
 *** 0230:            unit_exist = check_sd_unit(d, unit)
     0231:            add_sd_unit(d, unit, pkg, unit_exist)
     0232:            add_sd_user(d, unit.name, pkg)
     0233:        for name in listvar_to_list(d, 'SYSTEMD_ENVIRONMENT_FILE_%s' % pkg):
     0234:            add_env_file(d, name, pkg)
[... cut ...]
 *** 0117:        parser = systemd_parse_unit(d, path)
     0118:        inhibit = listvar_to_list(d, 'INHIBIT_SYSTEMD_RESTART_POLICY_WARNING')
     0119:        if unit.is_service and \
     0120:                not unit.is_template and \
     0121:                unit.name not in inhibit and \
File: '/home/vivek/gitlab/openBMC/openbmc/meta-phosphor/classes/obmc-phosphor-systemd.bbclass', lineno: 102, function: systemd_parse_unit
     0098:
     0099:
     0100:def systemd_parse_unit(d, path):

mahendra-space, 2025-04-23:
Please help on above issue which I am facing while building image for respberry pi openbmc

williamspatrick, 2025-04-23:
> Please help on above issue which I am facing while building image for respberry pi openbmc

Please do not comment on random already closed issues.

You are encountering a python issue but I have no idea:
a) what version of openbmc you are trying to run (hopefully master).
b) what OS you are trying to run on and what version of python that has.

The issue template for a new issue would ask you these questions.