openbmc/openbmc #3379: BMC goes unstable when BMC resets

gkeishin, 2018-09-17:
This is intermittent but has been seen lately in master build few times.
```
root@witherspoon:~# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.NotReady
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off
DBus error occurred: The name xyz.openbmc_project.State.Host was not provided by any .service files
```
```
root@witherspoon:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10
root@witherspoon:~# uptime
 13:25:50 up  3:12,  load average: 0.83, 1.13, 1.40
root@witherspoon:~# uname -a
Linux witherspoon 4.18.7-b7ba2a0af62b835fb3cac7bd3bbc72dcb1955b65 #1 Wed Sep 12 19:22:06 UTC 2018 armv6l GNU/Linux
```
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.3-332"
VERSION_ID="v2.3-332-g5f540a2"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.3-332"
BUILD_ID="v2.3"
root@witherspoon:~#
```
`Process top snapshot`
```
Mem: 265968K used, 163552K free, 1308K shrd, 13476K buff, 50072K cached
CPU:   0% usr 100% sys   0% nic   0% idle   0% io   0% irq   0% sirq
Load average: 19.99 10.67 4.39 17/137 2179
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1696     1 root     RN   99996  23%  43% /lib/systemd/systemd-coredump
  558     2 root     RW       0   0%  35% [ubi_bgt0d]
 2176  1108 root     R     3064   1%   9% top -n 1 -b
  734     1 root     S    22892   5%   7% /lib/systemd/systemd-journald
    1     0 root     R    35444   8%   2% /lib/systemd/systemd
 1853     1 root     S     6484   2%   2% phosphor-dump-manager
 1388     1 root     RN    4748   1%   2% /lib/systemd/systemd-coredump
 1328     1 root     RN   99996  23%   0% /lib/systemd/systemd-coredump
 1556     1 root     RN   99996  23%   0% /lib/systemd/systemd-coredump
 1392     1 root     DN   99996  23%   0% /lib/systemd/systemd-coredump
 1360     1 root     RN   99996  23%   0% /lib/systemd/systemd-coredump
 1698     1 root     RN   99996  23%   0% /lib/systemd/systemd-coredump
 2179     1 root     R    35444   8%   0% /lib/systemd/systemd
 1906     1 root     S    33808   8%   0% power_control.exe 500 10
 1903     1 root     S    33804   8%   0% button_reset.exe
 1904     1 root     S    33804   8%   0% button_power.exe
 1428     1 root     S    31600   7%   0% python /usr/sbin/phosphor-gevent rest_dbus --no-ssl
 1039     1 root     S    19424   5%   0% /usr/sbin/rsyslogd -n
 1134  1129 www-data S    15640   4%   0% nginx: worker process
 1129     1 root     S    15500   4%   0% nginx: master process nginx -p /var/volatile/nginx
 1012     1 systemd- S    15252   4%   0% /lib/systemd/systemd-timesyncd
 1282     1 root     S<   15028   3%   0% python /usr/sbin/phosphor-mapper --path_namespaces=/org/open_power /xyz/openbmc_project /org/openbmc /com/ibm --interface_namespaces=xyz.openbmc_project org.openbmc com.ibm org.open_power --blacklists= --interface_blacklists=
 1365     1 root     S    13136   3%   0% python /usr/sbin/chassis_control.py
 1850     1 root     S    13132   3%   0% python /usr/sbin/system_manager.py
 1870     1 root     S     8552   2%   0% ipmid
 1254     1 root     S     8408   2%   0% phosphor-dbus-monitor
 1798     1 root     S     7796   2%   0% openpower-update-manager
 1857     1 root     S     7776   2%   0% phosphor-image-updater
 1703     1 root     R     7016   2%   0% phosphor-inventory
 1692     1 root     S     6916   2%   0% ibm-log-manager
 1697     1 root     S     6568   2%   0% phosphor-settings-manager
 1233     1 root     S     6540   2%   0% phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@100/bmp280@77
 1029     1 root     S     6532   2%   0% phosphor-user-manager
 1289     1 root     S     6512   2%   0% witherspoon-psu-monitor --path=/sys/bus/i2c/devices/3-0068 --instance=1 --inventory=/system/chassis/motherboard/powersupply1 --num-history-records=120 --sync-gpio-path=/dev/gpiochip0 --sync-gpio-num=105
 1291     1 root     S     6512   2%   0% witherspoon-psu-monitor --path=/sys/bus/i2c/devices/3-0069 --instance=0 --inventory=/system/chassis/motherboard/powersupply0 --num-history-records=120 --sync-gpio-path=/dev/gpiochip0 --sync-gpio-num=105
 1263     1 root     S     6432   1%   0% phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@180/ir35221@71
 1262     1 root     S     6432   1%   0% phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@180/ir35221@70
[... 67 lines cut ...]
    3     2 root     IW       0   0%   0% [kworker/0:0-eve]
    9     2 root     SW       0   0%   0% [kdevtmpfs]
  209     2 root     SW       0   0%   0% [watchdogd]
 1382     2 root     IW       0   0%   0% [kworker/u2:2-ev]
 2178  2003 root     RW       0   0%   0% [basename]
    2     0 root     SW       0   0%   0% [kthreadd]
    4     2 root     IW<      0   0%   0% [kworker/0:0H-kb]
    6     2 root     IW<      0   0%   0% [mm_percpu_wq]
   87     2 root     SW       0   0%   0% [oom_reaper]
  185     2 root     IW<      0   0%   0% [writeback]
  187     2 root     IW<      0   0%   0% [crypto]
  189     2 root     IW<      0   0%   0% [kblockd]
  236     2 root     SW       0   0%   0% [kswapd0]
  518     2 root     IW<      0   0%   0% [ipv6_addrconf]
  564     2 root     SW       0   0%   0% [ubi_bgt4d]
 1047     2 root     IW       0   0%   0% [kworker/0:3-eve]
 1071     2 root     SW       0   0%   0% [ubi_bgt8d]
 1092     2 root     SW       0   0%   0% [ubifs_bgt8_1]
 1100     2 root     SW       0   0%   0% [ubifs_bgt8_2]
 1182     2 root     SW       0   0%   0% [ubifs_bgt4_0]
 1240     2 root     IW       0   0%   0% [kworker/0:4-eve]
 1457     2 root     SW       0   0%   0% [ubifs_bgt8_3]
```
- [x] Journald log 
[BMC_service_crashes.txt](https://github.com/openbmc/openbmc/files/2388789/BMC_service_crashes.txt)

geissonator, 2018-09-17:
https://github.com/openbmc/openbmc/issues/3369 is tracking the root issue here (which is still unknown). I'd like to use this bug to track the side affects of this bug. We should not have services endlessly restarting.

Because the timeout on a service default to 90s, and we allow up to 2 restarts before giving up, we need our StartLimitIntervalSec to be relative to this.

rfrandse, 2018-09-17:
<https://gerrit.openbmc-project.xyz/12874> Increase StartLimitIntervalSec to 240s
Resolves  #3379 BMC goes unstable on resets

geissonator, 2018-09-27:
Had to revert my change due to https://github.com/openbmc/openbmc/issues/3393, so reopening.

rfrandse, 2018-10-25:
<https://gerrit.openbmc-project.xyz/14745> Disable systemd-coredump from phosphor layer
resolves  #3379 BMC goes unstable on resets

closed 2018-10-30 by geissonator