openbmc/openbmc #3380: cross-localedef-native fails to build

shenki, 2018-09-19:
master on ppc64le Ubuntu 18.10:

```
ERROR: cross-localedef-native-2.27-r0 do_compile: oe_runmake failed
ERROR: cross-localedef-native-2.27-r0 do_compile: Function failed: do_compile (log file is located at /scratch/joel/openbmc/qemuarm/tmp/work/ppc64le-linux/cross-localedef-native/2.27-r0/temp/log.do_compile.66834)
ERROR: Logfile of failure stored in: /scratch/joel/openbmc/qemuarm/tmp/work/ppc64le-linux/cross-localedef-native/2.27-r0/temp/log.do_compile.66834
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-64', 'powerpc-common', 'common-linux', 'common-glibc', 'ppc64le-linux', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 128
| gcc  -o localedef charmap.o charmap-dir.o ld-address.o ld-collate.o ld-ctype.o ld-identification.o ld-measurement.o ld-messages.o ld-monetary.o ld-name.o ld-numeric.o ld-paper.o ld-telephone.o ld-time.o linereader.o localedef.o locarchive.o locfile.o record-status.o repertoire.o simple-hash.o xmalloc.o xstrdup.o argp-ba.o argp-eexst.o argp-fmtstream.o argp-help.o argp-parse.o argp-pv.o argp-xinl.o error.o getopt.o getopt1.o md5.o obstack.o asprintf.o getdelim.o localedef_extra.o obstack_printf.o vasprintf.o
| /scratch/joel/openbmc/qemuarm/tmp/hosttools/ld: argp-fmtstream.o: in function `_argp_fmtstream_update':
| argp-fmtstream.c:(.text+0x5a8): undefined reference to `_IO_fwide'
| /scratch/joel/openbmc/qemuarm/tmp/hosttools/ld: argp-fmtstream.c:(.text+0x614): undefined reference to `_IO_fwide'
| /scratch/joel/openbmc/qemuarm/tmp/hosttools/ld: argp-help.o: in function `argp_failure':
| argp-help.c:(.text+0x2f10): undefined reference to `_IO_fwide'
| collect2: error: ld returned 1 exit status
| make: *** [Makefile:60: localedef] Error 1
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /scratch/joel/openbmc/qemuarm/tmp/work/ppc64le-linux/cross-localedef-native/2.27-r0/temp/log.do_compile.66834)
ERROR: Task (/scratch/joel/openbmc/meta/recipes-core/glibc/cross-localedef-native_2.27.bb:do_compile) failed with exit code '1'
```

brando56894, 2018-11-07:
I'm trying to compile for aspeed and I get the same error

original-birdman, 2018-12-21:
Try adding, `-D_IO_fwide=fwide` to the CFLAGS additions in `openbmc/poky/meta/recipes-core/glibc/cross-localedef-native_2.27.bb`.

closed 2019-07-22 by stale[bot]